kliment/Printrun #609: config lost on app crash

bstott, 2015-01-01:
Where is the personal configuration file normally located for the application on Windows? I had my application settings configured and have been using Pronterface while occasionally making changes for the last full year. My OS crashed and on reboot the recent configuration was gone replaced with a default configuration. 

Is my recent changes saved somewhere or written over (lost?)?

OS: Win 8
Printrun: 1.2.3

iXce, 2015-01-01:
Users\yourusername\printrunconf.ini ?

bstott, 2015-01-01:
Curious. That file has not been changed since March 2014 and yet I have changed the settings several times since and saved to have them invoked on each time the application runs.

bstott, 2015-01-01:
Curious even more is my pronsolerc is mostly [null] entries where the colors, custom buttons, macro and other settings were. The *.bak file is the same.

Request/Suggestion: The backup file should be of a previous time - not related to crash - and secure. Maybe with a query to change on saving or writing over? I don't know really but, this did happen earlier this year (2014) - or last (2013) with my files on a win 7/Vista laptop. Today, 1/1/2015 was on a Win 8 OS and different computer from previous. Argh....