kliment/Printrun #610: Pronterface Serial Idles While GUI Not Focused For Indeterminate Duration

cdaringe, 2015-01-01:
Hi all:

This is a long standing issue (at least two years now) pertaining exclusively to usage while using OSX.

# Problem Statement
While printing over USB to a variety of firmwares, often, if the actual window GUI `focus` is changed to a different window for *some indeterminate amount of time*, the print idles.

Simply refocusing on the Pronterface gui kicks it back into gear.  It's as though the O/S determines that that the `pid` is ILDE and executes some power saving logic/etc.  I have had a difficult time repeatably reproducing the failure, however have observed it on two macbooks, one a 2010 air w/ snow leopard, and one a 2014 pro w/ mavericks.

It would be great if other OSX users could remark on their experience as well.

v3rm0n, 2015-01-04:
Same thing happens for me too on OSX Yosemite, Late 2011 Pro.

kliment, 2015-01-04:
-----BEGIN PGP SIGNED MESSAGE-----
Hash: SHA1

I think it's something to do with the OSX power management. See if
disabling appnap helps.

-----BEGIN PGP SIGNATURE-----
Version: GnuPG v1

iQIcBAEBAgAGBQJUqXZfAAoJEC9FHFwEZzqVv28QAKZF757lzao6U3UZmooPyleA
AAy4TEPxfQdMj4nbrfYDpHKCUidtJeH5+wyebmDuiqQf0MOSabQKwjAuvoAMnZAw
Jlo8vEMLOPqxRPY0NiNRMwrPwWKt9AKDgYq+zMLkAaNoa7OagPwCZjADqqH+xRVl
dpoRIkqg/mGr/Thr1sqNDySBrGCtm5Qifz+3m7r4SIOVz6ok7gtlKTashleGbH6V
9AZo93DriPQeBbIw1H8ay9CED1UKvS64WeW/3lYrgHhJY0srA6YFHUSjW7F/J1kb
Rn86mi2kVB/bVElukUt0/mD3XQmeja6Q4NNPoQluFXCt+EuqZAb0mo9zzNExLuZM
wLJ7+5B6gInKH5EZMU0I92d68213f7AHT3qRw7BGB65XMZhfdngDkvvDxZ8DfOD+
xzubWrhpuu61yZIxaAcGI4fpaBzxZafA1gnBQA2ha4FVzglViGmLXiES2EdvVjw3
D/H39Eeu+JZpxMAmqmLMBP/DCSm8XH/vWmPIeRnb+4vvLSG48/eZgKehx2kfRYk+
LhXOWbQDVOKmbd5b4BxF1o6PFy8nW5a+BhB/H7NZ6XUh/6kP5+xu5mRSAhFiwlwq
H4w8wmFfAMmFq0Z6Vz0E1wDWgsmfyroCugNbhjlys2LQAdwN89YjknJgoP4fMf1X
ZdYj00lmRrejwghy3VsX
=U1pT
-----END PGP SIGNATURE-----

cdaringe, 2015-01-05:
Although it may be a contributor, I don't *think?* App Nap is on Snow Leopard, thus couldn't alone be the sole source

kliment, 2015-02-03:
Can you see if you can reproduce using latest binary?

cdaringe, 2015-02-03:
hey @kliment, I'm hanging out on your side of the world for a couple months, but when I return, I will be happy to load up the latest and test it.  was there a change that you felt could have impacted this issue in the release?

kliment, 2015-02-03:
Yes, the current binary both inhibits power management while printing and has a manifest entry that should disable appnap.

cdaringe, 2015-02-03:
excellent!  yahooo!  I have always printed using [caffeine](http://lightheadsw.com/caffeine/) to stop my prints from dying while i was in another room.  this will make life a little simpler :+1:

koalazak, 2015-06-16:
Same problem here. Yosemit 10.10.3 Macbook Pro mid 2010. Lastest pre-compiled binary.
I solve the problem disabling "App Nap" only in the Pronterface app:
1. Rigth click on Printrun-Mac-03Feb2015.app
2. Click on 'Get Info'
3. Check "Prevent App Nap" checkbox

rockstorm101, 2017-12-20:
Hi @cdaringe, @v3rm0n and @koalazak, I know the issue is old but in case you are still using Printrun, could you please check whether this also happens in the latest release?

Thanks

rockstorm101, 2018-03-06:
I'm closing this issue due to a lack of feedback. Problems seems solved to me. Please feel free to reopen if needed.

closed 2018-03-06 by rockstorm101