kliment/Printrun #611: Home position

dbrb2, 2015-01-03:
On my previous build of pronterface, when when I clicked on the "XY" button at the centre of the GUI manual controls, the head was homed with G28

Now the command G0 X85.0 Y85.0 
is sent...which for me crashes the carriages (kossel delta printer) 

I guess what it is trying to do is centre the probe without changing the height? But in my case that would be G0 X0 Y0

I have set the build plate to circular in options...am I missing something else?

iXce, 2015-01-03:
Well if the center of your bed is at 0,0, then you should set X/Y offsets to -85/-85 in the build dimensions settings, I think !

dbrb2, 2015-01-03:
That's certainly done the trick. I wasn't sure if this was expected behaviour, since it was a change since....some point  in the past when that button used to issue a G28

So long as this is how it is supposed to work, then fine :-) Thanks for the quick reply

iXce, 2015-01-03:
Glad it solved the issue and sorry you had a crash :)

closed 2015-02-02 by kliment