kliment/Printrun #612: connect first time only - arduino rev.3

confrey59, 2015-01-18:
Hi everybody,
I'm using a Arduino Mega 2560 with RAMPS1.4; all work fine while compiling and uploading firmware with arduino IDE, even if I repeat the same process more time; when I try to connect pronterface to the printer (a Kossel), I can connect it the first time, next, after a disconnection, I can't connect anymore, getting the message 

Attempted to write invalid text to console, which could be due to an invalid baudrate

it have no sense, at first time, I connect and all work fine, after disconnecting I can't do it again, but if I try to upload the firmware again it works, so I think it's not a system problem;  this happens ony with Mega 560 rev.3, using a ch341 kernel module.

many thanks

confrey59 (dario)

Nuvadusmart, 2015-01-18:
a full power cycle, with both powersupply and usb disconnected, will fix it.

1. disconnect psu
2.disconnect usb
3.connect usb
4. connect psu
5. press connect in printrun

Sendt fra min iPad

confrey59, 2015-03-17:
thanks for response, and excuse me if I reply so late; I can't use that type of arduino with this software, I tried to connect and reconnect in that order, but not ever I can connect to printer, I think the result is random; and it's not tinkable to reboot a PC to connect safety, because of I can upload firmware using Arduino IDE while I can't connect anymore with Pronterface; is it possible I'm alone with this problem?

kliment, 2015-03-17:
I think I have an idea of what's going on. Can you test something for me:

Step 1: Attempt to connect at the WRONG baudrate
Step 2: Once that fails, attempt to connect again at the correct baud rate
Maybe your serial driver is failing to set the baud rate once it's already been set to the same value. I've seen that before in the version of pyserial that ships with mint.

kliment, 2018-03-23:
This was a pyserial problem that has since been fixed upstream.

closed 2018-03-23 by kliment