kliment/Printrun #613: Crash when using 3D view

dethegeek, 2015-01-22:
Hi

I'm trying pronterface under debian Jessie (beta 2).

When pronterface tries to generate the view it freezes and after 20 or 30 seconds the whole graphic environment crashes, probably due to a crash in the graphic adapter's driver (nouveau in my setup)

I tried all view modes. It only crashes when using 3d mode without the lighter 3D visualization.

I had enough time to see one of my cores running 100% of its computing power.

kliment, 2015-02-03:
Do you have similar behavior with any other opengl apps?

iXce, 2015-02-03:
This is msot likely the same as the other crash reports we have. There's an issue somewhere in the "heavy" 3D viz code, probably driver-related. I'll investigate this as soon as I get access to a test setup again ^^

dethegeek, 2015-02-03:
Hi,

I played a a a small amount of time with gzdoom, and used Openscad for a learning purpose. None of them crashed.

I know there are a big amount of logs displayed on a text console while the computer is laggy. This is related to the kernel the graphic driver and Xorg. I'll reproduce the bug and check my syslog.

dethegeek, 2015-02-15:
Hi

Wanted today to reproduce the bug and give a log. The bug does not occurs anymore. Maybe some recent update on my system solved the problem.

iXce, 2015-02-15:
Interesting ! So it could just have been a drivers bug, which were not happy with the arrays we use or something ?

dethegeek, 2015-02-15:
I also was trying Xen Hypervisor a few weeks ago; but I did not checked if it was running at the same time I ran printrun. I'll reinstall xen and give a try right now.

dethegeek, 2015-02-15:
I still cannot reproduce with xen and printrun in dom0. If needed for other similar issues, I share some of my current specs  : 

uname -a : Linux thinkpad-w510 3.16.0-4-amd64 #1 SMP Debian 3.16.7-ckt2-1 (2014-12-08) x86_64 GNU/Linux
nouveau v1.0.11-1

xen 4.4.1
dom0 has 4GB RAM (physical RAM : 8GB)  and 4 vCPU (physical CPU : i7 820QM)

kliment, 2018-03-23:
I'm closing this since it's apparently not reproducible anymore. If this changes please reopen.

closed 2018-03-23 by kliment