kliment/Printrun #614: Request: up to date compled versions

RayMJr, 2015-01-24:
Would it be possible to get a current compile of the project for both the Mac and windoze versions of Printrun?

Or would it be possible to get the recipe to do it myself?

Thank You!

RM

a4jp-com, 2015-01-25:
Yes please.

kliment, 2015-02-02:
I've just rebuilt the Windows and OS X versions. 

As for recipes, the Windows build is done with py2exe - there are build files in the repository, setup_win.py and release_windows.bat. These will work out of the box (as soon as you correct the path to your Python installation), but there are two changes I made in the source. First, I changed the default slicer command to point to the bundled slic3r distribution. Second, I changed the icon bitmap type from wx.BITMAP_TYPE_PNG to wx.BITMAP_TYPE_ICO in pronterface.py, because that's what gets embedded in the executable. 

On mac os X I use PyInstaller 2.1 placed in a directory on the same level as the Printrun directory. All the SetIcon commands are disabled in the source. The slicer path in settings has been set to point to a Slic3r living in Applications. I'll post the build script I use but it has some hardcoded paths in it you will probably need to change. It assumes the previous build has been put in ./pronterface/dist/pronterfacemac.app so it can copy the Plist file and some resources over.

closed 2015-02-02 by kliment

kliment, 2015-02-02:
Here is the build script and .spec file I use to build on the mac - I am not sure if it will work with recent mac os versions - I'm running this on osx 10.5 http://pastebin.com/xDm4Gqp0