kliment/Printrun #615: wxPython 3 support

hroncok, 2015-01-26:
Hi, wxPython 3 has reached Fedora rawhide (will be Fedora 22). As a packager of Printrun, my concern is that it works with wxPython 3.

Currently, Pronterface runs, but it has no console and some input fileds are not very usable.

[image]

This is copied form stderr:

```
** (pronterface.py:6144): WARNING **: Couldn't register with accessibility bus: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.
*** BUG ***
In pixman_region32_init_rect: Invalid rectangle passed
Set a breakpoint on '_pixman_log_error' to debug

*** BUG ***
In pixman_region32_init_rect: Invalid rectangle passed
Set a breakpoint on '_pixman_log_error' to debug

*** BUG ***
In pixman_region32_init_rect: Invalid rectangle passed
Set a breakpoint on '_pixman_log_error' to debug

*** BUG ***
In pixman_region32_init_rect: Invalid rectangle passed
Set a breakpoint on '_pixman_log_error' to debug

```

Plater seems fine.

I might be able to go and try fixing this, but I have never used wxPython before, so it might be better idea if some of the regular contributors looks into this. Let me know if I can buy you a beer or something, in case you make this work :)

Thanks a lot.

kliment, 2015-02-02:
The console is hidden - does it show correctly if you click the < button on the top right? As for the input fields, we'll have to make them wider due to the new layout. The only difficult one will be the print speed one as the slider is already cramped. Maybe we need to split it out on two rows. I'll look into this but I'll need you to test the layout changes as of course the controls look very different for me. Poke me on IRC to coordinate.

hroncok, 2015-02-02:
So I tested again with current master.

First, the console is hidden, but the button indicates it is not (the arrow points out, as in hide)

[image]

When I click the button, the console is not shown, but the direction of the arrow is reversed, so now I read is as "show".

[image]

And yes, indeed, the button shows the console. (I've also noticed the extremely wide gap between the console and the gcode preview area.)

[image]

I can also hide the left panel.

[image]

But when I show it again, the console is again gone.

[image]

hroncok, 2015-04-26:
Fedora 22 is already at Beta. I would love to see this fixed before the stable release (currently planned to 2015-05-26).

While the missing console is no longer an issue. I still have the problem we've discussed with @kliment on IRC https://churchyard.fedorapeople.org/printrun.ogv

(Note that the lagging is not the issue, it's only happening when recording the video, the issue is that the widgets are expanding even if I make the window smaller.)

Thanks for any help.

rockstorm101, 2016-03-22:
Currently I don't have a system running wxPython version 2.8, does this only happen with 3.0?

kliment, 2016-03-22:
Yes, it only happens with 3.0

claell, 2016-11-26:
Any updates on this? Did this got fixed?

lologarithm, 2017-02-22:
I just downloaded master and everything *seems* to be working fine (and no console errors)

petterreinholdtsen, 2017-10-01:
For the record, this issue is linked to from Debian bug https://bugs.debian.org/819745 .

rockstorm101, 2018-03-03:
Sadly, I can still reproduce this issue with Printrun 2.0.0rc2. Can you confirm @hroncok?

hroncok, 2018-03-20:
This behaves nicely for me with 2.0.0rc3.

rockstorm101, 2018-03-20:
I still can reproduce this issue at 5edf6aa. :(

[image]

However I'm investigating whether this is due to my setup. Because I see the little '^' and 'v' buttons instead of the '+' and '-' that everyone else is seeing.

kliment, 2018-03-21:
Yeah rockstorm, looks like you're on gtk2