kliment/Printrun #616: feature request (and contribution): support multiline g-code in custom buttons

MoonCactus, 2015-01-27:
Copy/pasting multi-line gcode sequence to custom buttons works, but only for the session since only the first line survives after a restart.

By adding one line (below), I was able to support. It is probably not beautiful enough, but it is nonetheless very useful, e.g. to calibrate a delta by sending sequences of movement and Z-probes
Note: the gcode commands must then be separated with a pipe so that reloading the buttons will not fail. Did I say it was a dirty fix?

def precmd(self, line):
>>>>HERE>>>        line= line.replace("|", "\n") # allow multiline GCODE button
        if line.upper().startswith("M114"):
            self.userm114 += 1
        (...)

Screenshot at the very end of http://www.tridimake.com/2015/01/github-howto-contribution-submit-patches.html

kliment, 2015-01-27:
-----BEGIN PGP SIGNED MESSAGE-----
Hash: SHA1

Any reason why you don't just make a macro and then trigger the macro
with the button?

-----BEGIN PGP SIGNATURE-----
Version: GnuPG v1

iQIcBAEBAgAGBQJUx0h7AAoJEC9FHFwEZzqV0FYQALCR41PjHmgJA7tMmJC+5qru
fs3F8wiKXjFIAoxt30EYbJSzI6YJGUykTO2nzcUcOI+en+PI4TDvn/PUUM8lOhGW
pYjuppcftBJ5oHuXe+EB7n4k9Niqyx5GP2oilewiDV3VKMyYWD465yRhcU6wekr5
ehjPX78C7Bt5m0rGeBecDc3TjpU8UxGXzwKjGPWgaMvmKKxxClXbuJZA7YMNlfkY
kP1H+nFmIiRdTu3RBEOgMeKD80bVPyiZogNHhkBZzyQk2+Xb45peIl3pbFHNQWay
c+HprKw7U9/kXpdyLrh5NUAE+55CjqrR1xO+KWuVw9ySiSblaa+aS78n3xK7WFh5
2tfq4Pipzoasl7vwyq5RYxaQL+dOnDy9iCznUF5xxgzbc1O0n9Ri1KUA42vVy/hT
7ytK5sGJluwMkPi6vHfnLI/fxwHjAofTORrjvEUaKQ/PZTiggabdw+XSCgL/lk5G
czIBvjlPFQdW0HH006K4QikNuLdhMbY9zQxCp29xTGEVf8CZ3bGLxY0LQW2oJupJ
wlKy4lQFp6Ne3hMNwJrtbBEACdWFk/wT8l/EbAoxgapp5fmr1LBYEmR+uV/ezIJq
LD09w+bxqAALR9BsqoMDWkH9mwWFrGEdjtGtVZ3C8tl7AFJeW3Tsf59B83HWmWnT
IdgDxeNo3JSb2c4UT5al
=N40+
-----END PGP SIGNATURE-----

MoonCactus, 2015-01-27:
Uh oh, I feel lonely suddenly! ;) Your solution is both cleaner and easier to edit.

Since I saw "add a custom button", I did not look further... I just felt it was restrictive that only one command was allowed so I dug further without stepping back and check if there was another way (the "bug" is where the button is reloaded but I had no time to check further).

Then, the command linked to a button could use the same multiline text box as the macros, and we would get the best of both worlds.

Anyway, sorry for this :)

hg42, 2016-01-24:
close?

MoonCactus, 2016-01-24:
Oh yes, sorry. I did not see I could close issues myself - done!

closed 2016-01-24 by MoonCactus