kliment/Printrun #617: motors and heater not operating

tronic71, 2015-01-31:
the sangulinolou is communicating with pronterface but command are ineffective to move motors or get heater to operate

tronic71, 2015-01-31:
also the buttons for x,y and z are not working

iXce, 2015-01-31:
Can you enable "Debug communications" checkbox and check what GCode
Pronterface sends ?

kliment, 2015-01-31:
-----BEGIN PGP SIGNED MESSAGE-----
Hash: SHA1

I know it seems obvious, but do you have 12V power attached to the board?

-----BEGIN PGP SIGNATURE-----
Version: GnuPG v1

iQIcBAEBAgAGBQJUzVxmAAoJEC9FHFwEZzqVdJgP/jMolSDWEMyeGYODzq9kPdCw
A/bGB8th5a8SxFC2GSmFvZZKKC0xSNPxZK0pTelKPUVnzcUkpKMZu4ig0HKlvsBY
zftQ7ZwJ+kchImLRlgVQt14ZuioRHURzW5obK+rZWv8n0DpcGLUxY6cxpKH7idDr
gXKFfK5exUt+w++WkyHMqEXst7ZpFHnN05ghtGYTa3yRnmsBUMt/cBlqJRSGp9sI
kzjEZsnamGdAsoOgiJ1E4pkfjNJygDUmpyoN2pxoPimTk7FJZX3w7uO3drixl/Fc
uMn6Bgt47XQs+9xlxntjI4TPPALvlZu8UqAEUXBw1nKljnep3sLcH9ou5xPeor2c
WUrhyblOE2TxOWDOiw2wlktx9DOUud225RTFcSjyAEQ707QJZPevFS/Rvu/bB1Le
GlJ0o7otUNjj2uYtPgYu2Jt+R+mpjbTZnQ0O3YOoAlJvY4dGmdq1DMjgArO+l2RU
8atE1j3avongUowu3s3JVSp+yYo8LVjujzETilQPLMgAGz5CHiA8zcmMpAVIqJ6N
OywIQ+jj3KBDECpMyCy2NlbMRqvSkJq5c5UHU2fTj1ee57ItfwKTvA70atzIjSz7
eKRTw6IIxzu8qfQhiaqay3JLJt+R8zaVLhUXnFzij3v/VrCxWpVGjeCCMBkMMJ5D
Kfns8M5lQSIvps+G636E
=vKAy
-----END PGP SIGNATURE-----

tronic71, 2015-02-01:
yes the 12volt supply is 11.9 volts and is connected. The green led is lit.
As I have only just got started on 3d printing I find the Pronterface strange in that it only seems to have OFF buttons for motors and heater. I tried sending a few commands to move motors and put on the heater and although the codes seem to be sent nothing happens on the Prusai3
Today I will recheck all of the cables and look at the debug communications
I really appreciate your quick responses, I really am keen to print something as well as calibrate the Prusai3.  Initially I had a lot of problems getting the boot into the Sangulinololu, but after that the Marlin software compiled and loaded ok.  The Pronterface did nothing after pressing the connect button, but after pressing the reset button on the board it communicated and gave the impression all was well.

kliment, 2015-02-01:
-----BEGIN PGP SIGNED MESSAGE-----
Hash: SHA1

Okay, two things. First, can you check that you have the correct
communication speed set (Marlin default is 250000)? Second, can you
post a copy of the communications in debug mode so we can see what's
going on.

To activate the heaters, enter the temperature you want to set them
to, and press "set"

-----BEGIN PGP SIGNATURE-----
Version: GnuPG v1

iQIcBAEBAgAGBQJUzgT2AAoJEC9FHFwEZzqVIJQP/2E+ImJeTkT+HtwT5t6cQw+l
lAYq0yldikH4waF0BhB061Cs37NhT8XVVYdjkzAN2vkZ5Pw3VqA6FPI7/HXwYZ93
PnSzIftdK0Qtq2CLqS+idF9DXROBGblmzJnElkTo1fncHKD7n1GtEF+dTFIEEmR/
aPiQ6kuJpVSnZm+Em0+zbhS3i219n6FJpKcWSkZ8OYT7TxJetUzLPoZcdMY11ePY
Uuw0HKCzqeeKTTXdZIcesKCxF5DK+ZE/mJZ025l9Ir1pwK1ZVnMJ4Yp2eSHUJiGW
AEs15clqmHAPGfpZ0d3jV3ya1QMgsvy0uvlRedYlWwjGMf8nYm/2HaB8mwZr35rZ
NzLlxMMtH06gT5NNP9xTbSvxl7Oa/qody35zVJtXbmjCbD322QWJ81v/z+zTQdNp
dnquV2SjYlavB3kwts/7CznEzJ74geKBynUty7RcNOeAiZxloAQuVVXaDs7f5jZ9
0+gyriujytWbGOKqJGKDTT6kNmck+jRcW5tGq4AyIl+r3PE6txRYhT890gFPNe9n
iDlFaDIZopMH8/+HaKVizqTeZGoBDA/gZpQrcJvG5YvtqMqZzo7xTjpvWVKhh4Hn
yYMqFTSDPt98iXuUUIGK2g2Z7FiuV6EFbhuVEbZOpjWszpnS8RAbByWKOphvahnK
AP+m/XrvOFXRjL4O/Qnk
=K8Dp
-----END PGP SIGNATURE-----

tronic71, 2015-02-01:
Hi
I have saved the communicates between the mac and the prusai3

file as follows

Regards Mike Baker

hairykiwi, 2015-02-01:
Mike, I couldn't see any attachment. You might try just copy-pasting the comms text you see in pronterface straight into your reply here. And/or grab a screenshot or two and share them here directly or on a photo sharing site.

tronic71, 2015-02-02:
Hi
Pasted here

Connecting...
start
Printer is now online.
echo: External Rese
Setting hotend temperature to 185.000000 degrees Celsius.
Setting bed temperature to 60.000000 degrees Celsius.
SENDING:M84
SENDING:M84
SENDING:M84
SENDING:M84
SENDING:M84
Slicing /Volumes/data1/Hobby/3D Printing/mc pig/pig.stl
Slicing /Applications/Slic3r.app/Contents/MacOS/slic3r /Volumes/data1/Hobby/3D Printing/mc pig/pig.stl --output /Volumes/data1/Hobby/3D Printing/mc pig/pig_export.gcode
=> Processing triangulated mesh
=> Generating perimeters
=> Preparing infill
=> Infilling layers
=> Generating skirt/brim
=> Exporting G-code to /Volumes/data1/Hobby/3D Printing/mc pig/pig_export.gcode
Done. Process took 0 minutes and 6.230 seconds
Filament required: 10333.7mm (73.0cm3)
Loaded /Volumes/data1/Hobby/3D Printing/mc pig/pig_export.gcode, 180667 lines
10333.7298188 mm of filament used in this print
The print goes:
- from 73.49 mm to 126.51 mm in X and is 53.03 mm wide
- from 52.25 mm to 154.51 mm in Y and is 102.26 mm deep
- from 0.00 mm to 63.95 mm in Z and is 63.95 mm high
Estimated duration: 2:46:01
Print started at: 16:59:46

An hour later nothing had heated or moved.
Regards
Mike Baker

> On 1 Feb 2015, at 18:20, tronic71co <tronic71@calatron.co.uk> wrote:
> 
> <Prusai3 comms.rtf>

kliment, 2015-02-02:
-----BEGIN PGP SIGNED MESSAGE-----
Hash: SHA1

Okay, looks promising on the communication front. Can you do the same
in debug mode so we can see what the printer sends back?

-----BEGIN PGP SIGNATURE-----
Version: GnuPG v1

iQIcBAEBAgAGBQJUzz9UAAoJEC9FHFwEZzqVqIAP/3WMktxQEqtEjgCdVZAYYUGM
hZNzN5cCQqiVuG/4aupo0O9rhJOH3XA0ubdl+S48AK2sO7pPzj4YiezrzknzErVt
UhF2BowMw+g9xtO7jnUH1aEOHDr3jna4mbrh7PiS+cHFXlcz3yf2mDzNSQ3mqkD2
MqRsmwfcMKP59Xbh8XbS+V5f0Dw8y872rLWyLjXzuxVyoqzNWtBDJpaTpWCf+nRv
Z+aYmjn16ea49AH2aHCGbTM9xCDgybrorz+vEnMeZaTBnJz2zAumbkxWs92Ygpqw
NyHCVor/F8trxons0KXyktbkt+v+GVH53X5kLZ4mD2Ee2r5cXjABZ2Po7QPxbqGe
3mxbUsLgif+XzoXu01dYkn9lM6dDZ4WNKS1bWWAJPyGVm6H5oTvN65ivegdUhOXW
rs4baF6rZbTrx3Jwsa2VNUAPqpbE7nsqLEYNe5AdqyY7fjt4YTGYRIwaDyY01eR3
TehQ1j+7tqqerONJRaf/lq0cmSQ/ASEeKfliz6H+GyvezmLDUk6jTzydiaZeaq8n
NxqUUjmT2nwYAn3QDhEgwQBKmHqx6/C82MpnCRHPt2aIEw3y10JuEYiA9jD02mPI
DQg6vmf5xD3RJWyz3QDTIMqd7Se0zqUWPDF9Q+N7K8IdVzJuHg4p4WLLF/tYnV4D
iqf+qHmxrVt/0hhAQnWE
=a+k6
-----END PGP SIGNATURE-----

dlarue, 2015-02-02:
Does the Arduino power up without the USB cable plugged in( look for the LEDs )? If you have the LCD display connected that'll tell you too.  It sounds like the 12V power supply is not supplying much power. Another thing to look at would be the 12V line after setting the hotend temp to 180C.

tronic71, 2015-02-03:
Hi
Yes it does power the Sangulinolo. The 12volt supply is the same whatever is selected but anyway I don’t get any heat  at all

I have tried other software on the mac to drive the marlin/prusai3 however nothing happy the same, so I have concluded that the problem us with the firmware.  I am going to replace it but have to redo the boot loader because it won’t talk to the Arduino at 250000 baud.
Thanks for your assistance.  I will let you know how I get on.
Regards
Mike Baker

kliment, 2015-02-03:
Okay. Closing this for now because it doesn't appear to be a problem with Printrun

closed 2015-02-03 by kliment