kliment/Printrun #619: Printrun-Mac-03Feb2015.zip - odd exception when printing

mhackney, 2015-02-05:
I just installed the new Printrun-Mac-03Feb2015.zip on OS X 10.10.2 (Yosemite) this morning. Everything configured fine. I am commenting to a smoothieboard over tcp to a 3D printer. When I load the gcode file, all looks good. Then when I run, I get this in the console:

Print started at: 09:44:22
[ERROR] Failed to set power settings:
Traceback (most recent call last):
  File "/Users/craigvanvliet/Desktop/PyInstaller-2.1/pronterface/build/pronterface/out00-PYZ.pyz/printrun.pronsole", line 1173, in startcb
  File "/Users/craigvanvliet/Desktop/PyInstaller-2.1/pronterface/build/pronterface/out00-PYZ.pyz/printrun.power", line 130, in powerset_print_start
  File "/Users/craigvanvliet/Desktop/PyInstaller-2.1/pronterface/build/pronterface/out00-PYZ.pyz/printrun.power.osx", line 72, in inhibit_sleep_osx
  File "/Users/craigvanvliet/Desktop/PyInstaller-2.1/pronterface/build/pronterface/out00-PYZ.pyz/printrun.power.osx", line 49, in AssertionCreateWithName
  File "/Users/craigvanvliet/Desktop/PyInstaller-2.1/pronterface/build/pronterface/out00-PYZ.pyz/printrun.power.osx", line 40, in StringToCFString
AttributeError: 'module' object has no attribute 'kCFStringEncodingASCII'

The object starts printing and then I get more errors thrown to console:

[ERROR] Can't write to printer (disconnected ?):
Traceback (most recent call last):
  File "/Users/craigvanvliet/Desktop/PyInstaller-2.1/pronterface/build/pronterface/out00-PYZ.pyz/printrun.printcore", line 608, in _send
  File "/Users/craigvanvliet/Desktop/PyInstaller-2.1/pronterface/build/pronterface/out00-PYZ.pyz/socket", line 324, in write
  File "/Users/craigvanvliet/Desktop/PyInstaller-2.1/pronterface/build/pronterface/out00-PYZ.pyz/socket", line 303, in flush
timeout: timed out

[ERROR] Can't write to printer (disconnected ?):
Traceback (most recent call last):
  File "/Users/craigvanvliet/Desktop/PyInstaller-2.1/pronterface/build/pronterface/out00-PYZ.pyz/printrun.printcore", line 608, in _send
  File "/Users/craigvanvliet/Desktop/PyInstaller-2.1/pronterface/build/pronterface/out00-PYZ.pyz/socket", line 324, in write
  File "/Users/craigvanvliet/Desktop/PyInstaller-2.1/pronterface/build/pronterface/out00-PYZ.pyz/socket", line 303, in flush
timeout: timed out

[ERROR] Can't write to printer (disconnected ?):
Traceback (most recent call last):
  File "/Users/craigvanvliet/Desktop/PyInstaller-2.1/pronterface/build/pronterface/out00-PYZ.pyz/printrun.printcore", line 608, in _send
  File "/Users/craigvanvliet/Desktop/PyInstaller-2.1/pronterface/build/pronterface/out00-PYZ.pyz/socket", line 324, in write
  File "/Users/craigvanvliet/Desktop/PyInstaller-2.1/pronterface/build/pronterface/out00-PYZ.pyz/socket", line 303, in flush
timeout: timed out

[ERROR] Can't write to printer (disconnected ?):
Traceback (most recent call last):
  File "/Users/craigvanvliet/Desktop/PyInstaller-2.1/pronterface/build/pronterface/out00-PYZ.pyz/printrun.printcore", line 608, in _send
  File "/Users/craigvanvliet/Desktop/PyInstaller-2.1/pronterface/build/pronterface/out00-PYZ.pyz/socket", line 324, in write
  File "/Users/craigvanvliet/Desktop/PyInstaller-2.1/pronterface/build/pronterface/out00-PYZ.pyz/socket", line 303, in flush
timeout: timed out

[ERROR] Can't write to printer (disconnected ?):
Traceback (most recent call last):
  File "/Users/craigvanvliet/Desktop/PyInstaller-2.1/pronterface/build/pronterface/out00-PYZ.pyz/printrun.printcore", line 608, in _send
  File "/Users/craigvanvliet/Desktop/PyInstaller-2.1/pronterface/build/pronterface/out00-PYZ.pyz/socket", line 324, in write
  File "/Users/craigvanvliet/Desktop/PyInstaller-2.1/pronterface/build/pronterface/out00-PYZ.pyz/socket", line 303, in flush
timeout: timed out

[ERROR] Can't write to printer (disconnected ?):
Traceback (most recent call last):
  File "/Users/craigvanvliet/Desktop/PyInstaller-2.1/pronterface/build/pronterface/out00-PYZ.pyz/printrun.printcore", line 608, in _send
  File "/Users/craigvanvliet/Desktop/PyInstaller-2.1/pronterface/build/pronterface/out00-PYZ.pyz/socket", line 324, in write
  File "/Users/craigvanvliet/Desktop/PyInstaller-2.1/pronterface/build/pronterface/out00-PYZ.pyz/socket", line 303, in flush
timeout: timed out

[ERROR] Disconnecting after 4 failed writes.
Disconnected.
[ERROR] Can't write to printer (disconnected ?):
Traceback (most recent call last):
  File "/Users/craigvanvliet/Desktop/PyInstaller-2.1/pronterface/build/pronterface/out00-PYZ.pyz/printrun.printcore", line 608, in _send
  File "/Users/craigvanvliet/Desktop/PyInstaller-2.1/pronterface/build/pronterface/out00-PYZ.pyz/socket", line 324, in write
  File "/Users/craigvanvliet/Desktop/PyInstaller-2.1/pronterface/build/pronterface/out00-PYZ.pyz/socket", line 303, in flush
timeout: timed out

[ERROR] Failed to set power settings:
Traceback (most recent call last):
  File "/Users/craigvanvliet/Desktop/PyInstaller-2.1/pronterface/build/pronterface/out00-PYZ.pyz/printrun.pronsole", line 1180, in endcb
  File "/Users/craigvanvliet/Desktop/PyInstaller-2.1/pronterface/build/pronterface/out00-PYZ.pyz/printrun.power", line 134, in powerset_print_stop
  File "/Users/craigvanvliet/Desktop/PyInstaller-2.1/pronterface/build/pronterface/out00-PYZ.pyz/printrun.power.osx", line 79, in deinhibit_sleep_osx
AttributeError: 'function' object has no attribute 'assertion_id'

It is printing all through the above. It seems odd that there would be a path to /Users/craigvanvliet - my user name is mhackney on my mac, Craig must be a developer?

mhackney, 2015-02-05:
Just an update. The printer/Pronterface lock up and must be reset. This happened on 3 prints somewhere around the 3rd to 4th layers. I have not been able to complete a print. When this happened, there were no errors in the console (they were long after the string of errors I attached above and I cleared console so I could look for new errors - there were none, just the lock up).

kliment, 2015-02-05:
It's the username of someone who kindly donated a mac for these builds - pyinstaller embeds the full path into the file.

The timeout is likely due to network errors - might need to adjust the timeout value there, the tcp printing is not particularly extensively tested. Strange that the print didn't stop after the "disconnected" message, by all measures it should have. I'm not entirely sure what's going on here.

The other error is means we don't have a means of inhibiting power management. It's probably a namespace issue due to the way pyobjc does dynamic import. I'll see if I can do anything about it.

mhackney, 2015-02-05:
Thanks for the update. I don't think I can change the timeout without compiling the code can I?

I have the smoothie direct to a 5 port gigabit ethernet switch. My mac host is on another port. the switch direct connects to my cable modem. All traffic should be from my mac to the printer. the smoothie port does show an amber light meaning it is not a gigabit connection.

mhackney, 2015-02-13:
An update klimant, the print actually does stop after the disconnect, its just that there is gcode buffered at the other end so it takes some time before the print stops, but it does. 

I am still experiencing this problem 100% of the time. I tested on 2 different networks with completely different cables, etc. I've monitored the network and see no issues with latency. I have a similar issue with timeouts when using the Python upload script that's now part of the Smoothieboard project. I had to increase the timeout to 20sec. I've had lots of issues with my Smoothieboard so it very well could be a hardware issue with it. I am getting another board soon so I can test when that comes in.

sunomie, 2015-02-20:
I can confirm also seeing '[ERROR] Failed to set power settings' when attempting to print from OSX.

fearfulWizard, 2015-02-20:
I am also having the same issue.. Does it have to do with the build?

kliment, 2015-02-20:
-----BEGIN PGP SIGNED MESSAGE-----
Hash: SHA1

Does the actual print work or does it stop after that error?

-----BEGIN PGP SIGNATURE-----
Version: GnuPG v1

iQIcBAEBAgAGBQJU5w2YAAoJEC9FHFwEZzqVGAQP/iTqwKPI80W7bwVMyXPKpZ58
ifsvFGr6FIZZacezrd3R9chs1aOW89YHY4uWoA5Nx0A9PlfhuKWS1N1hrV6e2uuu
xQjVsPR3v7eLuo7aIaFKfnoK0lINhxvIffhWaAwhcY0AWMEUPddiD4fGd8TmNeYF
Ap/K+l0zwLdmsckm0mG45CShbqCiIZv9DQMaNtkWbsAEZl5wg65sDXVbY/yOYlJh
eE1adGX1+Y+k6IKyNiuGA+QKOq8iHDAF4OlBYqIzgHZeeVrf1sy3ZMqkfy7ONhHM
jjstxhNbhau4J3ui4UxOKd9rnG15TEJvUjsRM49D9+v1SfeRNK6wn6E2BpJ07zTi
R7X14Gud+n5pvbEK9e81tkRWZawulC7a1uqqTxLqQew0plLi5jlYzUkSzuk0Zikk
diBdLSewm2DZsJXsdV3Ox5tTSDOyuMn8Q3QJdk7L/ZuEJU6dzyY5gZ65jwcFFikf
oPNiicd6CI+fNnEM+S18jR7ZGNc04oYEmS4+6jMSQkq+OOrGq5YiQCBbdibkLVFZ
E9rT3jll20E1FAHOaRMo5YpnlwjTctzZ0gdZZVR3Ay21BbEO9vJEHV2hVG0dvfmH
JCVQjx2hJjswZMdnHhrGqmNRDMAcMHlDmHk5CCv6WBO7P3tp5eUZ/HZGLk2i8TS7
enSEqDqmuFxfDH/IjH8t
=hulT
-----END PGP SIGNATURE-----

mhackney, 2015-02-20:
For me, after I get the [ERROR] Failed to set power settings: message, the print starts but then fails shortly with a Not Connected message. This is 100% reproducible and I have not been able to use this version to print.

iXce, 2015-02-20:
It's most likely unrelated to the error message, which is solely related to some power settings helper and not to the rest of the print stack. Now we need to trace these disconnections :/

fearfulWizard, 2015-02-20:
The printer does not work after that error. I must mention though, I do not have the extruder yet on my printer, so I tried simply running it by having it draw a 2D image with a pencil. This is my first printer, so i might've missed the "power settings", but I'm not sure what this is referring to: the configuration of the firmware, or power settings of the Gcode.

iXce, 2015-02-20:
It actually refers to power settings of the host operating system, that is to make Printrun run as high priority and inhibit sleep. So it shouldn't affect the print itself.

sunomie, 2015-02-22:
Does not start printing for me on OSX. Ended up switching to my Ubuntu box
and compiling then printing.

fearfulWizard, 2015-02-23:
I think I've managed to solve this temporarily by switching to a previous build. Please do let me know how I can fix the issue if it's mainly a namespace error. Thanks

mhackney, 2015-02-24:
Another update - the Windows version running in a Win 7 VM on the same mac is working fine, no network time outs. And of course it does not generate that initial "Failed to set power settings"

malemodel, 2015-04-15:
so?! is craig vanvliet on this list?! awesomeness...

kliment, 2015-04-15:
craig donated the mac computer that I use to make the mac builds. I'll
take that path out of the build script for the next one to avoid confusion

malemodel, 2015-04-15:
cool. great. when i that happening?

mjbouchery, 2015-05-02:
Printer is now online.
echo:Marlin: 1.0.0 RC2
echo: Last Updated: Jan 30 2015 19:44:56 | Author: nophead
echo: Free Memory: 12982  PlannerBufferBytes: 1232
echo:Using Default settings:
echo:SD card ok
Loading file: /Users/mjohn/Desktop/Cl1.gco
Loaded /Users/mjohn/Desktop/Cl1.gco, 40009 lines
277.10mm of filament used in this print
The print goes:
- from 0.00 mm to 113.10 mm in X and is 113.10 mm wide
- from 0.00 mm to 126.88 mm in Y and is 126.88 mm deep
- from 0.00 mm to 23.10 mm in Z and is 23.10 mm high
Estimated duration: 228 layers, 1:07:56
echo:SD card ok
Writing to file: Cl1.g
Print started at: 16:02:33
[ERROR] Failed to set power settings:
Traceback (most recent call last):
  File "/Users/craigvanvliet/Desktop/PyInstaller-2.1/pronterface/build/pronterface/out00-PYZ.pyz/printrun.pronsole", line 1173, in startcb
  File "/Users/craigvanvliet/Desktop/PyInstaller-2.1/pronterface/build/pronterface/out00-PYZ.pyz/printrun.power", line 130, in powerset_print_start
  File "/Users/craigvanvliet/Desktop/PyInstaller-2.1/pronterface/build/pronterface/out00-PYZ.pyz/printrun.power.osx", line 72, in inhibit_sleep_osx
  File "/Users/craigvanvliet/Desktop/PyInstaller-2.1/pronterface/build/pronterface/out00-PYZ.pyz/printrun.power.osx", line 49, in AssertionCreateWithName
  File "/Users/craigvanvliet/Desktop/PyInstaller-2.1/pronterface/build/pronterface/out00-PYZ.pyz/printrun.power.osx", line 40, in StringToCFString
AttributeError: 'module' object has no attribute 'kCFStringEncodingASCII'Printer is now online.

I printed the the android test little guy with minor issues (printer was not calibrated yet) but i downloaded the kendal90 slic3r configuration that I got from here. 

http://wiki.imal.org/howto/how-print-mendel-90

opened up a part I was going to try and print exported gcode and I got that in pronterface.

b4stien, 2015-05-06:
Exact same problem here, with a gcod file as well.

mjbouchery, 2015-05-11:
I wonder what is going on. It is weird. It printed the android guy just fine. but with the other gcode I am having an issue.

JohnWasser, 2015-05-14:
Same problem (Mac OS X Yosemite 10.10.3):
Print started at: 10:35:52
[ERROR] Failed to set power settings:
Traceback (most recent call last):
...
  File "/Users/craigvanvliet/Desktop/PyInstaller-2.1/pronterface/build/pronterface/out00-PYZ.pyz/printrun.power", line 130, in powerset_print_start
  File "/Users/craigvanvliet/Desktop/PyInstaller-2.1/pronterface/build/pronterface/out00-PYZ.pyz/printrun.power.osx", line 72, in inhibit_sleep_osx
  File "/Users/craigvanvliet/Desktop/PyInstaller-2.1/pronterface/build/pronterface/out00-PYZ.pyz/printrun.power.osx", line 49, in AssertionCreateWithName
  File "/Users/craigvanvliet/Desktop/PyInstaller-2.1/pronterface/build/pronterface/out00-PYZ.pyz/printrun.power.osx", line 40, in StringToCFString
AttributeError: 'module' object has no attribute 'kCFStringEncodingASCII'

JohnWasser, 2015-05-14:
Looks to me like in printrun/power/osx.py lines 40m and 41 should be changed from:
```
    encoding = CoreFoundation.kCFStringEncodingASCII
    cfstring = CoreFoundation.CFStringCreateWithCString(None, string, encoding);
```
to:
```
    // encoding = CoreFoundation.kCFStringEncodingASCII
    cfstring = CoreFoundation.CFStringCreateWithCString(None, string, kCFStringEncodingASCII)
```

iXce, 2015-05-14:
Hmm, but kCFStringEncodingASCII does not exist in this context, does it ?

JohnWasser, 2015-05-14:
iXce, I expect it is a constant defined when you import CoreFoundation.

iXce, 2015-05-14:
That's not how things work in Python I think, if nothing explicitly imports the symbol it cannot be there (you'd need something like from blah import * for something like this to work)

kliment, 2015-05-14:
None of the CoreFoundation stuff makes any sense to me honestly, so I'm
willing to try it out.

iXce, 2015-05-14:
Note that // is not a valid way to comment things in python either, so I still highly doubt this works :P

iXce, 2015-05-14:
Let's debug this properly: could you pop a python shell, do "import CoreFoundation" and then "print dir(CoreFoundation)" ? Thanks !

JohnWasser, 2015-05-14:
Python 2.7.6 (default, Sep  9 2014, 15:04:36) 
[GCC 4.2.1 Compatible Apple LLVM 6.0 (clang-600.0.39)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import CoreFoundation
>>> dir(CoreFoundation)
['AEVConversionUtils', 'AppleBluetoothHIDDevice', 'AtherosHostController', 'AutoCropper', 'BNEPControl', 'BTPrivilegedMachBootstrapServer', 'BluetoothHIDDevice', 'BluetoothHIDDeviceController', 'BluetoothUIServer', 'BroadcomHostController', 'CAAdditionalCertInfo', 'CAAnimation', 'CAAnimationGroup', 'CABackdropLayer', 'CABasicAnimation', 'CABasicConstraintsExtension', 'CABehavior', 'CABoxLayoutManager', 'CACGPathCodingProxy', 'CACGPathCodingSegment', 'CACGPatternCodingProxy', 'CACertInfo', 'CACodingProxy', 'CAConstraint', 'CAConstraintLayoutManager', 'CAContext', 'CAContextImpl', 'CADebuggingArchiverDelegate', 'CADistanceFieldLayer', 'CADynamicsBehavior', 'CAEmitterBehavior', 'CAEmitterCell', 'CAEmitterLayer', 'CAExtendedKeyUsageExtension', 'CAFilter', 'CAForceField', 'CAGradientLayer', 'CAIOSurfaceCodingProxy', 'CAIdentityName', 'CAKeyPairAttributes', 'CAKeyUsageExtension', 'CAKeyframeAnimation', 'CALayer', 'CALayerArray', 'CALayerHost', 'CALight', 'CALinearMaskLayer', 'CAMLParser', 'CAMLWriter', 'CAMatchMoveAnimation', 'CAMediaTimingFunction', 'CAMediaTimingFunctionBuiltin', 'CAMeshInterpolator', 'CAMeshTransform', 'CAMutableMeshTransform', 'CAOpenGLLayer', 'CAPackage', 'CAPluginLayer', 'CAPropertyAnimation', 'CAProxyLayer', 'CARemoteLayerClient', 'CARemoteLayerServer', 'CARenderer', 'CAReplicatorLayer', 'CAScrollLayer', 'CAScrollLayoutManager', 'CAShapeLayer', 'CASlotProxy', 'CASmoothedTextLayer', 'CASpring', 'CASpringAnimation', 'CAState', 'CAStateAddAnimation', 'CAStateAddElement', 'CAStateController', 'CAStateControllerAnimation', 'CAStateControllerLayer', 'CAStateControllerTransition', 'CAStateControllerUndo', 'CAStateElement', 'CAStateRemoveAnimation', 'CAStateRemoveElement', 'CAStateSetValue', 'CAStateTransition', 'CAStateTransitionElement', 'CASubjectAltNameExtension', 'CASublayerEnumerator', 'CATableLayoutManager', 'CATextLayer', 'CATiledLayer', 'CATransaction', 'CATransactionCompletionItem', 'CATransformLayer', 'CATransition', 'CAValueFunction', 'CAWrappedLayoutManager', 'CBATTRequest', 'CBCentral', 'CBCentralManager', 'CBCharacteristic', 'CBDescriptor', 'CBMutableCharacteristic', 'CBMutableDescriptor', 'CBMutableService', 'CBPeripheral', 'CBPeripheralManager', 'CBService', 'CBUUID', 'CBXpcConnection', 'CDImageComponent', 'CDImageData', 'CFAbsoluteTimeAddGregorianUnits', 'CFAbsoluteTimeGetCurrent', 'CFAbsoluteTimeGetDayOfWeek', 'CFAbsoluteTimeGetDayOfYear', 'CFAbsoluteTimeGetDifferenceAsGregorianUnits', 'CFAbsoluteTimeGetGregorianDate', 'CFAbsoluteTimeGetWeekOfYear', 'CFAllocatorGetDefault', 'CFAllocatorGetPreferredSizeForSize', 'CFAllocatorGetTypeID', 'CFAllocatorRef', 'CFAllocatorSetDefault', 'CFArrayAppendArray', 'CFArrayAppendValue', 'CFArrayApplyFunction', 'CFArrayBSearchValues', 'CFArrayContainsValue', 'CFArrayCreate', 'CFArrayCreateCopy', 'CFArrayCreateMutable', 'CFArrayCreateMutableCopy', 'CFArrayExchangeValuesAtIndices', 'CFArrayGetCount', 'CFArrayGetCountOfValue', 'CFArrayGetFirstIndexOfValue', 'CFArrayGetLastIndexOfValue', 'CFArrayGetTypeID', 'CFArrayGetValueAtIndex', 'CFArrayGetValues', 'CFArrayInsertValueAtIndex', 'CFArrayRef', 'CFArrayRemoveAllValues', 'CFArrayRemoveValueAtIndex', 'CFArrayReplaceValues', 'CFArraySetValueAtIndex', 'CFArraySortValues', 'CFAttributedStringBeginEditing', 'CFAttributedStringCreate', 'CFAttributedStringCreateCopy', 'CFAttributedStringCreateMutable', 'CFAttributedStringCreateMutableCopy', 'CFAttributedStringCreateWithSubstring', 'CFAttributedStringEndEditing', 'CFAttributedStringGetAttribute', 'CFAttributedStringGetAttributeAndLongestEffectiveRange', 'CFAttributedStringGetAttributes', 'CFAttributedStringGetAttributesAndLongestEffectiveRange', 'CFAttributedStringGetLength', 'CFAttributedStringGetMutableString', 'CFAttributedStringGetString', 'CFAttributedStringGetTypeID', 'CFAttributedStringRef', 'CFAttributedStringRemoveAttribute', 'CFAttributedStringReplaceAttributedString', 'CFAttributedStringReplaceString', 'CFAttributedStringSetAttribute', 'CFAttributedStringSetAttributes', 'CFBagAddValue', 'CFBagApplyFunction', 'CFBagContainsValue', 'CFBagCreate', 'CFBagCreateCopy', 'CFBagCreateMutable', 'CFBagCreateMutableCopy', 'CFBagGetCount', 'CFBagGetCountOfValue', 'CFBagGetTypeID', 'CFBagGetValue', 'CFBagGetValueIfPresent', 'CFBagGetValues', 'CFBagRef', 'CFBagRemoveAllValues', 'CFBagRemoveValue', 'CFBagReplaceValue', 'CFBagSetValue', 'CFBinaryHeapAddValue', 'CFBinaryHeapApplyFunction', 'CFBinaryHeapContainsValue', 'CFBinaryHeapCreate', 'CFBinaryHeapCreateCopy', 'CFBinaryHeapGetCount', 'CFBinaryHeapGetCountOfValue', 'CFBinaryHeapGetMinimum', 'CFBinaryHeapGetMinimumIfPresent', 'CFBinaryHeapGetTypeID', 'CFBinaryHeapGetValues', 'CFBinaryHeapRef', 'CFBinaryHeapRemoveAllValues', 'CFBinaryHeapRemoveMinimumValue', 'CFBitVectorContainsBit', 'CFBitVectorCreate', 'CFBitVectorCreateCopy', 'CFBitVectorCreateMutable', 'CFBitVectorCreateMutableCopy', 'CFBitVectorFlipBitAtIndex', 'CFBitVectorFlipBits', 'CFBitVectorGetBitAtIndex', 'CFBitVectorGetBits', 'CFBitVectorGetCount', 'CFBitVectorGetCountOfBit', 'CFBitVectorGetFirstIndexOfBit', 'CFBitVectorGetLastIndexOfBit', 'CFBitVectorGetTypeID', 'CFBitVectorRef', 'CFBitVectorSetAllBits', 'CFBitVectorSetBitAtIndex', 'CFBitVectorSetBits', 'CFBitVectorSetCount', 'CFBooleanGetTypeID', 'CFBooleanGetVa [... cut ...] eText', 'kCFXMLNodeTypeWhitespace', 'kCFXMLParserAddImpliedAttributes', 'kCFXMLParserAllOptions', 'kCFXMLParserNoOptions', 'kCFXMLParserReplacePhysicalEntities', 'kCFXMLParserResolveExternalEntities', 'kCFXMLParserSkipMetaData', 'kCFXMLParserSkipWhitespace', 'kCFXMLParserValidateDocument', 'kCFXMLStatusParseInProgress', 'kCFXMLStatusParseNotBegun', 'kCFXMLStatusParseSuccessful', 'kCFXMLTreeErrorDescription', 'kCFXMLTreeErrorLineNumber', 'kCFXMLTreeErrorLocation', 'kCFXMLTreeErrorStatusCode']
>>>

JohnWasser, 2015-05-14:
Typing CoreFoundation.kCFStringEncodingASCII into the python shell displays 1536 (= 0x600 which is how it is defined in CFString.h)

iXce, 2015-05-14:
Okai so it's definitely in the system CoreFoundation module. Now the question is, why isn't it in the binary ? Is a different CoreFoundation module shipped with the binary ?
I pushed a dirty workaround for now

kliment, 2015-05-14:
It's in there but when corefoundation is packed into a binary everything
gets renamed in some weird way

mjbouchery, 2015-05-15:
I am running pronterface on my macbook pro just an FYI

clos8, 2015-05-16:
Is it working for you with no issues?

aboobed, 2015-06-26:
من القائمة Apple، اختر "تفضيلات النظام ..." و من تفضيلات النظام انقر على "الأمن
[image]

Thank you for all ..baltokid talking about this

sderose, 2015-11-12:
Did this ever get resolved? I'm seeing the same problem here, and I don't see a newer version of PrintRun to try....

mjbouchery, 2015-11-12:
I have been so busy with a new born and work I haven't had a chance to try
anything. It's odd.

markwbrown, 2016-06-17:
This is over a year old, yet the fix has not been made. Is anyone on this?

MacBert, 2016-06-18:
Hey Markwbrown, I'm only a very begging user with 3D printing, and I like Printrun, but this is a killer problem, I'm not a programer so I can't help solve it, but if I can run tests for the Programing Gurus, I'm more than willing.

kliment, 2018-03-27:
Can someone test with the 2.0.0rc5 release?

allthemoose, 2018-04-20:
Hi Kliment, I just installed the 2.0.0rc5 (pronterface) and have the same issue. The error message has changed to read 
Print started at: 00:28:00
Failed to set power settings:
Traceback (most recent call last):
  File "printrun/pronsole.py", line 1219, in startcb
  File "printrun/power/__init__.py", line 130, in powerset_print_start
  File "printrun/power/osx.py", line 75, in inhibit_sleep_osx
  File "printrun/power/osx.py", line 52, in AssertionCreateWithName
  File "printrun/power/osx.py", line 44, in StringToCFString
ValueError: Expecting byte string of length 1, got a 'str'

allthemoose, 2018-04-20:
Just noticed when I put my Gcode file directly into the downloads folder this error doesnt appear. I previously had it under /Downloads/10MM_Test_CUBE_WITH_SKIRT_CALIBRATION_TEST/files/ maybe the underscores?

closed 2018-05-11 by kliment