kliment/Printrun #620: Bug while openning GCodes

Kermareg, 2015-02-12:
Hello,
It seems I've problems while opening differents GCodes files and the threes got a problem.
Two of them are realized with inkscape and gcodetools, the third one is created with Cura (Ive tried with Cura V14 et Cura V15.
If I open the Cura's one, I got a python stacktrace:
>Loading file: /xxx/xxx/testImprimante3DKermareg.gcode
>Exception in thread Thread-3:
>Traceback (most recent call last):
>  File "/usr/lib/python2.7/threading.py", line 552, in __bootstrap_inner
>    self.run()
>  File "/usr/lib/python2.7/threading.py", line 505, in run
>    self.__target(*self.__args, **self.__kwargs)
>  File "/home/tesla/Programmes/Printrun/printrun/pronterface.py", line 1368, in load_gcode_async_thread
>    gcode = gcode)
>  File "/home/tesla/Programmes/Printrun/printrun/pronsole.py", line 888, in load_gcode
>    layer_callback = layer_callback)
>  File "/home/tesla/Programmes/Printrun/printrun/gcoder.py", line 222, in prepare
>    layer_callback = layer_callback)
>  File "/home/tesla/Programmes/Printrun/printrun/gcoder.py", line 410, in _preprocess
>    current_tool = int(line.command[1:])
>ValueError: invalid literal for int() with base 10: ''

If I try to open one created with inkscape, but this time, parsing seems "correct" until the summary of the file says 
>Loading file: /xxx/xxx/triskell.gcode
>Loaded /xxx/xxx/triskell.gcode, 2017 lines
>0.00mm of filament used in this print
>The print goes:
>- from 0.00 mm to 0.00 mm in X and is 0.00 mm wide
>- from 0.00 mm to 0.00 mm in Y and is 0.00 mm deep
>- from 0.00 mm to 0.00 mm in Z and is 0.00 mm high
>Estimated duration: 0 layers, 0:00:00

Also, there's nothing in the preview field. I just have the first layer for the Cura generated file but for the two inkscape's one, there's nothing.
Bizarre, I can launch a print even if there's "0 layers". It only start the first path and after nothing move.

As I can't upload here the 3 files, they're uploaded at [my personnal website](http://kermareg.be/gcode/).
If you want to translate your software in other language, I can help you for the french translation.

Thanks.
Kermareg

kliment, 2015-02-12:
-----BEGIN PGP SIGNED MESSAGE-----
Hash: SHA1

I can't seem to open the files there - I get connection timeouts

-----BEGIN PGP SIGNATURE-----
Version: GnuPG v1

iQIcBAEBAgAGBQJU3RkPAAoJEC9FHFwEZzqVTroP/jXEQLOiEFV/Xe4Ee8wb0c20
F8uuLjx/q8Pu3L8K0vBWuwWbykOzsMFLJXcadBnSRBXxdqloKY8JegUUPA7KUPyP
IdoT6vaA6VTRwebc1RzMi2eB67n67Pi9Nbp6VOcPtCARvEe6cjPjH4V49qBldcPo
mmDZmrk5bGXfTs1+sHGExgfQ1i/9cdesY28gIikL2K09EGFC+h+tiAGcFbQbet6/
pbuz808mGxQVgztLc9ZScV2HxCUDIo2eXdv2HnhZWTSfReOFdqJXcFJyrTtytLwV
b/2iv+3busaJskR5+rcwE8qdZMGEDgrW2fgJY+mxVl98NWKKi8EV1iRVjviWZatf
TkkM9//zDuwmGipJiPrvX7bXDq+PF8MobTvowSsCsI+GgeFgWjRJlGPwSQCe7SQ4
NPEhXxu6xE2wXMzQ692FSv4qIffwFfzI1+fBx4hSIOv8KItN9+ZqrIHc63SHVQMv
LfrLz5H74p1OGJOuglo/nAJlKfrN3saOlmMBqC/advDh4acNiLyaUT8HA4CYEg7R
vufnj4iE9sbcZICtQBcgHpQdgqoFA9ryHONk89T6d9DGLZ2qFMUgDa3DEBpRP59T
Oa6PPxCDb2GM7TFWyQMWvdUjmSbH2jtI6ImvibhDUwdBfq5jFnebpZJJ9OIwc6RL
uB9d7YgffVIs493uPzBF
=V9lP
-----END PGP SIGNATURE-----

iXce, 2015-02-12:
Could only download the triskell one for now. I *think* it's most likely failing on this file because the commands are spelled with an extra 0 (G01 instead of G1), etc. Is this standard ?

The other fail seems to be (I think) because of a "T" command without any argument. I was not aware this was possible, if it's in the gcode spec we should totally work on supporting this.

Kermareg, 2015-02-12:
I got some errors too, I'll send you by mail if you inform me. Sorry for the inconvenience.

Kermareg, 2015-02-13:
@kliment and @iXce the web issue seems to be corrected.
I'll try by removing the unused 0.

Kermareg, 2015-02-13:
Ok, I'm testing on triskell.gcode. 
I've removed all the optional 0 in the file. Super, I've the plot on the preview area. Still one problem: I got always a freeze at the first letter. I have any stack trace, just the communication trace:
>[...]
SENT: N24 G1 X100.239670 Y19.692290 Z-1.000000*97
SENT: N24 G1 X100.239670 Y19.692290 Z-1.000000*97
RECV: ok
RECV: ok
SENT: M105
SENT: M105
RECV: ok T:54 B:126
RECV: ok T:54 B:126
SENT: N25 G1 X98.977388 Y19.692290 Z-1.000000*83
SENT: N25 G1 X98.977388 Y19.692290 Z-1.000000*83
RECV: ok
RECV: ok
SENT: N26 G1 X98.977388 Y17.107089 Z-1.000000*95
SENT: N26 G1 X98.977388 Y17.107089 Z-1.000000*95
RECV: ok
RECV: ok
SENT: N27 G3 X99.012071 Y16.882587 Z-1.000000 I0.743865 J-0.000013*123
SENT: N27 G3 X99.012071 Y16.882587 Z-1.000000 I0.743865 J-0.000013*123

If I reinit the board, i got a checksum mismatch?
>Réinitialisée.
SENT: M105
SENT: M105
SENT: M105
RECV: Error: checksum mismatch, Last Line:26
RECV: Error: checksum mismatch, Last Line:26
[ERROR] Error: checksum mismatch, Last Line:26
[ERROR] Error: checksum mismatch, Last Line:26
RECV: Resend:27
RECV: Resend:27
RECV: ok
RECV: ok
SENT: M105
[...]

If I relaunch a print, it freezes at the same point
Thanks all for your help.

kliment, 2015-02-13:
-----BEGIN PGP SIGNED MESSAGE-----
Hash: SHA1

Not corrected for me - can you put them on pastebin?

-----BEGIN PGP SIGNATURE-----
Version: GnuPG v1

iQIcBAEBAgAGBQJU3cGKAAoJEC9FHFwEZzqVkToP/2RYl4MtWwrHdrxYzZytNNeb
r97yL5EfF2yXuheirRIih61OcdbRFv+llcu4sVP/v96UEl4tq+BOtWPCUgbRizDa
G+RTy7E4kzHktOClMC3CeOda7ocTfeYtsH7KXYNAxTC+7DSzBHIXIMbJ8IVfhidP
hQ7WUP73SXpKKYUw6WHdGxgg1Z7MP4k9WeGutlOP/TCuE2e8Zt6KeNBNLKesVgBY
4GDksM9FhP18nnjJXIcajPqpEmZLTLcYhLez5i88j7k6+sz4iEqTWNN4+Xvt8zNN
8wUXijKewoCEQ6o9MYqUL943UhJ6/vToGJmN/tAixdXLkza/uYb9XyuuFjMf9zMu
YS52/Cmvm0ShU1OG2ikRJDimJ3JYX2wVYGjbnrmH7x0SWvgxsnrJPpcGcfz35YMc
cNZIz5GrmHexreUXmzvFS1BH4M/sTwiEPud6ClTJFIbTOo7ugVtA795iHdz8ZP4o
t+tQrvQDr9sXVoGOhpQsBdofBBxl2M1vF3OVUAmvNtHFsJGgzwXmUg27M4jsC4VO
f9af7Zz+kEOyQeXgKy41OHQIdYaJ5FjuaD+S0S38lVxNHA6OVWw8kzxZHj16IcP5
HvdjikJcZOlyHTlU9RTMPNkPMM5OPYSwhNPD5BDDf+TH39PMRSz9BctBOZh4bO4r
ebgokZ1yLPN2FdbtNPw4
=Oho+
-----END PGP SIGNATURE-----

Kermareg, 2015-02-13:
You can find it [here](http://pastebin.com/w3XrafsN).
Do you have timeout or what else of errors on the domain ?

iXce, 2015-02-13:
Hmm, on my system, with my printer simulator, the gcode you pasted on pastebin prints fine. Based on your communication trace, I'd suspect that your machine does not support arcs (G3) ?

Kermareg, 2015-02-13:
I've the arc function enabled. And I don't know if G3 is in cause: some G3 functions are well executed like this one "SENT: N17 G3 X98.044695 Y20.706874 Z-1.000000 I-37.648391 J12.075920*119"
I've tried with a "normal' firmware which is unmodified (jsute the mother board) and I've the same issue.
I'm asking me if there is no problem with the *XXX at the end of your sending frame ? Indeed, they're not in my file.
Which firmware are you using ?

kliment, 2015-02-13:
-----BEGIN PGP SIGNED MESSAGE-----
Hash: SHA1

Those are checksums, and the printer should recognize them fine. They
are automatically calculated, and are not in the file. However, the
incidence of * or N in the file can confuse the checksum calculator.

-----BEGIN PGP SIGNATURE-----
Version: GnuPG v1

iQIcBAEBAgAGBQJU3cyUAAoJEC9FHFwEZzqVfbwQAN1e9AHEV81aYUsdVAVfAVNC
eDUqxPLSqnAu80IwLWX3/WGy2dV0+MOWm7jzT8ShUUYb+HNggI3KBLi6ILQN2qGK
OhMX0Dvg2R96yx6c65BxKKsrMUXEyjZJ+2PIzB5VKv/BJgKtHF1RIrchD5Ov/XLs
4W5pXwY0DJ0o3pwi2EmtEJwU5PFJ03wGkle8hNZEu4AVkM50aGMilqBUsLXzVfpe
8jfgsnSyDD6oPnk/Wvb3gJ5cYynzK6Xr/Cf57+mnHMmBXhDNxrisoBuldEjOHpk4
3UyNvjFQl1ihRkNwJJ6o79o/1lEY6Z6KSdrEv0Sc0h2loo1GBCLEwV11SC5YitDB
8z4FO1rRrO3jTdXuqLFqkl2OXlIMnXMAPcEndI0XFOAdGmd4b1fapsthkd/74c81
dfYHKWsqbns8Tbs6aluukVc7hIPQ/Wl7b2zEf9NCsYMWYxRkqv78RDunId8Ae+KB
rX3yML/xgnP+omUuyff72tc0pZ0ewmxL/y7252NekNeFl6jMCB0uv0ctu3Wi6s6K
C0x4fOUY1IyUTk4k3idqRROgo7FP87+h2+tQcnDSJOJ3HxrOtpRoDooCw4a3UeBr
VHrShBQSztYkxFTrK10ezvPcNdp9/oNjCoUm0hHy+L4fGkfXl4t50K+faCkfiYwX
VCCcXdwZZH/X6dIYGtBm
=fQPJ
-----END PGP SIGNATURE-----

Kermareg, 2015-02-13:
I've also tested with a windows built (if the problem is coming from my os) and its the same. 
>SENT: N25 G1 X98.977388 Y19.692290 Z-1.000000*83
RECV: ok
SENT: N26 G1 X98.977388 Y17.107089 Z-1.000000*95
RECV: ok
SENT: N27 G3 X99.012071 Y16.882587 Z-1.000000 I0.743865 J-0.000013*123
Print paused at: 13:03:54
SENT: M105
RECV: Error: checksum mismatch, Last Line:26
[ERROR] Error: checksum mismatch, Last Line:26
RECV: Resend:27
RECV: ok

If I try to read/"print" via EMC, there's no issues in the code. 
Can you check the checksum's computation ? Is there a way to disable properly the checksum check to skirt the problem ?

kliment, 2015-02-13:
-----BEGIN PGP SIGNED MESSAGE-----
Hash: SHA1

Checksum calculation is correct - it's the xor-sum of each character,
in this case N27 G3 X99.012071 Y16.882587 Z-1.000000 I0.743865
J-0.000013 adds up to 123, as expected. Sounds like your firmware is
failing to calculate said checksum correctly. What firmware are you using?

-----BEGIN PGP SIGNATURE-----
Version: GnuPG v1

iQIcBAEBAgAGBQJU3fB7AAoJEC9FHFwEZzqVVgoQAI6vFguyZFz7jeXD+6Kpa0A9
l+HNFbvNCmGjqjo3ccnWRW6G8oXgvuAj8WMu1Gi2A9kGmayAePJdhq6lRL5aWDZ3
7i7okZ74DLIj2whKcZUqat1jtPETBgkzt44d1s25bZlMSUbpU8XVZuA29SownUuV
AtsjzH8mQdEvnxz9bCmV8LvmDqfxzeV6KXIjnW5i1+ABdRsPN+xwOkReVBzHicDZ
dmOck0qcvjkkyF1Gln27vlhIesS0mzEejSrXAWFSmd5tLQtld0NQgMGrB1GQwwrC
4lUdQByZ5FsCKJTKf/CxqdwnEtRXxzeKZw+f6qNa509+6Es0F71Y4MMRI1qtMUIV
ZR185fkr9nQ0AhUh6MysWmW6NrGCvpHbob7UZ6EMooBAZS0ikwBqrrsRMBy7Nqz4
Ol6Nz//HYaAJG3V5izAcU8lslsApTCnScn1mhFhXTExwODfnsC3sFSUY0bi60i8r
OeeoHTNYmXhBUHvps/2uc2oQRDsD8QOUlaKCDp+HFfVLTpuL0o7Nc7Mflg8j/4Uu
Cm8hpoIiLJWxV+1d4F1O53/JWxKH8/kAnGcvJMSepAeTTRaCuQvVn0VbqmiRmrXv
6ZGe52Y1FORXXJG0xUYQIFfv7iqHQEjKoYDOTVltfoGYjF7s394ulG5fWY4hyhHm
bRH05FYxL8AW7g/JyFAj
=Hcvb
-----END PGP SIGNATURE-----

Kermareg, 2015-02-13:
I'm using Sprinter V1.3.04T.
I said before that I'm using it with some modifications, but the firmware used now is without modification. So it's "normal" :smile: 
I think a found something: if I raise down the resolution manualy (only 3 decimals) the program continue untill the lines become to 6 decimals.

kliment, 2015-02-13:
-----BEGIN PGP SIGNED MESSAGE-----
Hash: SHA1

You're probably hitting the maximum line length in the firmware - you
can increase that and see if it helps

-----BEGIN PGP SIGNATURE-----
Version: GnuPG v1

iQIcBAEBAgAGBQJU3fPaAAoJEC9FHFwEZzqV1JsQAKK1usChCGD85htJ5HiKLzTv
O1En4e/JZxAYm+auLQrgX/h8YaUFGgmwQ5gVTV6iB1s4vG9KvSjkzPkKCZi+9t8J
kWpbO02i2mz9cQC8mdSwWfXdHgwo9OuUk6mZN+GNosBZF+FE1Gefmv2AGStOwjvR
z7gKwdpeot8ZmjQq4KcF3ZhYXXX7a+j0b8sfY/PwEQuH8cqxdMMQieqls91ujVcE
COTaPsY/Ulkh3Xs5FC2Dywq229yNDSFR3MyRtU5OOU0SKGYLY/JyBErBQmr7EKRh
dDhXkWBSLfW6Ijb/JFfCw4v88lHlbds7MyjlvkoCtlpkQLtn6riGpisZlDB7c/OO
QqSCVxWxCKNG9416Dw5QUU+61MS9zTkPB9LHSITU3nF+yYF1bXNQMAPI+z9u7gGJ
K+XdYeY6QI/W1Z1hlznZR1GbNr3pbFBzxY0kR2j4VtWani9/GnxKdB/LXMtkUpwA
dWvIEuw3NSc9W0EslcHXbcuzdVZv65XM4LSACo95shpP63SleytEtT0j0e5A//nR
am1OXfmyI0MD1tUkOxZ37JTJ0NNn681X1mEJf19bK6LBjgfORwddJT00w4xrUfpL
B6zsNwi1Z4EHq3RFcOMD9UtaZtrIq0PQfMh9koCOAcCrgpD5v4QLfpTVhMf0/EgJ
k9kn4u7FZP8u8RMbv79I
=B6Pr
-----END PGP SIGNATURE-----

iXce, 2015-02-13:
That's definitely what I thought when I saw the file (lines too long because the precision is high) but the weird thing is that there are some longer lines before that one..

Kermareg, 2015-02-13:
@iXce yep, and the big question is why these lines are well parsed and other no ?
@kliment can you tell me how can I increase the size in the firmware ? it seems that it's not in Sprinter.h nor configuration.h

iXce, 2015-02-13:
@Kermareg: maybe because they are so long that the checksum character is ignored, while on that 27th command it's just the right size to make it fail ? (though I thought that if using N.. the firmware should expect the checksum and fail if not getting one)

kliment, 2015-02-13:
-----BEGIN PGP SIGNED MESSAGE-----
Hash: SHA1

On 02/13/2015 02:00 PM, Kermareg wrote:
> @kliment can you tell me how can I increase the size in the
> firmware ? it seems that it's not in Sprinter.h nor
> configuration.h

https://github.com/kliment/Sprinter/blob/master/Sprinter/Sprinter.pde#L349
-----BEGIN PGP SIGNATURE-----
Version: GnuPG v1

iQIcBAEBAgAGBQJU3fZAAAoJEC9FHFwEZzqV6vEP/RgLBbokw/8caL4tF1jd6g7K
8eCnHL8Rka+ETbgGJ059JigS6d2R2q32xAkOdfZuBrk/Xh1uWzh3CC3l7CYLvOvJ
9Gq2hOlMTphpu+5SdGFtPNW/4edRylZ82gLMmvT5LAffz8sSvjYOU0JrqN2nf4UZ
pYqlBBRjZFEfxPidMgfT8sU5Q9YkaUPeyjxYruQfZXV4EE7AF8XtOQJ2ls7ISRyW
6ZYlToBVCNNDSMAsrzuX2zBxlFvRX/piTca+jbArVngzc7ygEUmH4DZEpA8zZy+U
faSqL6rIKdIzbwIjeMa46069xd7A1pwCLQcxpR8ahsawqr7ZvMhvKo6BgWnj3irF
G/glVSGI6NbcaFHWZ03TcggCMxN+3E6/oslMxW9GiePtdKFnrxpbo8F+M98shRmy
hbDCykNIdsVhGNgiu9jYyX6brn0IoXnp8HlsB5xQFFdm+bnw9JWYupgJ17fIr1FJ
OrK5zlPSbvHI5YXuam2vFBVnVI+CVU6dzmdSwAPq7cyrhV5jTI3/JsnTa7Pj+gc7
/OL5tKOZ8N0PRZ2dAjfE8YI7lkX3AUMCC924KewqOFEWZx73gCTFcebLX7nH24V6
nGDAyfma+M4HJioYHIXt2XdRj4iK8P2kuBHg8epxCYhLFKZISbSeF4fZIdvYXcLH
JpMBUa7rigno+ariFFgN
=UEUf
-----END PGP SIGNATURE-----

Kermareg, 2015-02-13:
Ok, thanks.
I've increased to 256 and the buffre from 6 to 10 but still the same.
I'll try to make a routine to modify the decimals as I cant't choose it in the CAM. I'll say if it's a solution or not.

Kermareg, 2015-02-13:
Ok Guys, 
I've created a mini routine to truncate the 6 decimals to 3 and... it works fine.
So the problem is coming from the parsing of hyper precise gcode which cause some errors in the checksum verification.
So in conclution:
* Avoid the usage of G01, G02 but use G1, G2,...
* Avoid creating a gcode file with too much decimals. 3 decimals gives a precision in the order of the µm.

Merci beaucoup pour le support, thanks a lot for the support.

Regards.

closed 2015-02-13 by Kermareg

iXce, 2015-02-13:
Glad it works for you now ! However I still wonder why it was not working, if it's not the firmware buffer, what is it ? We definitely send exactly what's written in the communication log, but maybe the serial driver is not receving it ? it would be interesting to run "echo" sketch which just replies ok + the command which was sent, or some statistics on it

Kermareg, 2015-02-13:
What do you think about a buffer overflow in the crc computation in the arduino side ? Maybe the sum is too big and make a rollover of the register. I've not looked carefully the source code but it could be there. 
As you say, a simple echo sketch to compare crc could be interesting. 
If I've time I can look a little bit but not this weekend, I'll be away.