kliment/Printrun #621: Cannot build from source for Windows

gege2b, 2015-02-13:
Hi all

I wish to build printrun from source, I've read the #614 issue wich explain how to do, but I got some problems anyway.

I'm using Win7 x64 with python27, and Running pronterface.py works fine

First, I had to modify the setup_win.py to add "import py2exe" and change the name of "P-face.ico" to "pronterface.ico"

Well, that was the first problem, quickly solved (may be it should be corrected on the repo ?)

When I run the release_windows.bat, it seems to work fine (byte-compiling eveything, copying everything, etc...) except the bundled slic3r but it's normal because I use it in standalone ^_^

But at the end, he tells me that some modules appears to be missing...

Here is the log : http://pastebin.com/VPmitbKd (I just replaced the base path with a dot)

When I try to run the resulting .exe, nothing happens, but the windows event log tell me that pronterface.exe has crashed (see log below)
If you want, I can upload the library.zip somewhere

I really wish to build pronterface myself, just to have the more up to date version on a lot of machine, without the need to install python and all the requirements

thanks

Here's the (useless ?) log :

    Version=1
    EventType=APPCRASH
    EventTime=130683081904613618
    ReportType=2
    Consent=1
    UploadTime=130683081905013618
    ReportIdentifier=50b93073-b385-11e4-a3bb-0090f5bc404c
    IntegratorReportIdentifier=50b93072-b385-11e4-a3bb-0090f5bc404c
    WOW64=1
    Response.BucketId=3098699055
    Response.BucketTable=1
    Response.type=4
    Sig[0].Name=Nom de l’application
    Sig[0].Value=pronterface.exe
    Sig[1].Name=Version de l’application
    Sig[1].Value=0.0.0.0
    Sig[2].Name=Horodatage de l’application
    Sig[2].Value=49180193
    Sig[3].Name=Nom du module par défaut
    Sig[3].Value=pronterface.exe
    Sig[4].Name=Version du module par défaut
    Sig[4].Value=0.0.0.0
    Sig[5].Name=Horodateur du module par défaut
    Sig[5].Value=49180193
    Sig[6].Name=Code de l’exception
    Sig[6].Value=c0000005
    Sig[7].Name=Décalage de l’exception
    Sig[7].Value=00002123
    DynamicSig[1].Name=Version du système
    DynamicSig[1].Value=6.1.7601.2.1.0.256.1
    DynamicSig[2].Name=Identificateur de paramètres régionaux
    DynamicSig[2].Value=1036
    DynamicSig[22].Name=Information supplémentaire n° 1
    DynamicSig[22].Value=0a9e
    DynamicSig[23].Name=Information supplémentaire n° 2
    DynamicSig[23].Value=0a9e372d3b4ad19135b953a78882e789
    DynamicSig[24].Name=Information supplémentaire n° 3
    DynamicSig[24].Value=0a9e
    DynamicSig[25].Name=Information supplémentaire n° 4
    DynamicSig[25].Value=0a9e372d3b4ad19135b953a78882e789
    UI[2]=.\dist\pronterface.exe
    UI[3]=pronterface.exe a cessé de fonctionner
    UI[4]=Windows peut rechercher une solution au problème en ligne.
    UI[5]=Rechercher une solution en ligne et fermer le programme
    UI[6]=Rechercher ultérieurement une solution en ligne et fermer le programme
    UI[7]=Fermer le programme
    LoadedModule[0]=.\dist\pronterface.exe
    LoadedModule[1]=C:\Windows\SysWOW64\ntdll.dll
    LoadedModule[2]=C:\Windows\syswow64\kernel32.dll
    LoadedModule[3]=C:\Windows\syswow64\KERNELBASE.dll
    LoadedModule[4]=C:\Windows\syswow64\USER32.dll
    LoadedModule[5]=C:\Windows\syswow64\GDI32.dll
    LoadedModule[6]=C:\Windows\syswow64\LPK.dll
    LoadedModule[7]=C:\Windows\syswow64\USP10.dll
    LoadedModule[8]=C:\Windows\syswow64\msvcrt.dll
    LoadedModule[9]=C:\Windows\syswow64\ADVAPI32.dll
    LoadedModule[10]=C:\Windows\SysWOW64\sechost.dll
    LoadedModule[11]=C:\Windows\syswow64\RPCRT4.dll
    LoadedModule[12]=C:\Windows\syswow64\SspiCli.dll
    LoadedModule[13]=C:\Windows\syswow64\CRYPTBASE.dll
    LoadedModule[14]=C:\Windows\WinSxS\x86_microsoft.vc90.crt_1fc8b3b9a1e18e3b_9.0.30729.6161_none_50934f2ebcb7eb57\MSVCR90.dll
    LoadedModule[15]=C:\Windows\system32\IMM32.DLL
    LoadedModule[16]=C:\Windows\syswow64\MSCTF.dll
    LoadedModule[17]=C:\Windows\syswow64\SHELL32.dll
    LoadedModule[18]=C:\Windows\syswow64\SHLWAPI.dll
    LoadedModule[19]=C:\Windows\syswow64\ole32.dll
    State[0].Key=Transport.DoneStage1
    State[0].Value=1
    State[1].Key=DataRequest
    State[1].Value=Bucket=-1196268241/nBucketTable=1/nResponse=1/n
    FriendlyEventName=Fonctionnement arrêté
    ConsentKey=APPCRASH
    AppName=pronterface.exe
    AppPath=.\dist\pronterface.exe

kliment, 2015-02-13:
-----BEGIN PGP SIGNED MESSAGE-----
Hash: SHA1

Unfortunately I've never managed a successful build on a 64 bit
Windows setup. If you figure out how to do it, or why it's not
working, please let me know. I do my builds on a 32 bit setup. The
missing modules should be no problem, everything looks about the same
as on mine. Just out of curiosity, does pronsole work correctly?

> Hi all
>
> I wish to build printrun from source, I've read the #614 issue wich
> explain how to do, but I got some problems anyway.
>
> I'm using Win7 x64 with python27, and Running pronterface.py works [...]
LoadedModule[15]=C:\Windows\system32\IMM32.DLL
> LoadedModule[16]=C:\Windows\syswow64\MSCTF.dll
> LoadedModule[17]=C:\Windows\syswow64\SHELL32.dll
> LoadedModule[18]=C:\Windows\syswow64\SHLWAPI.dll
> LoadedModule[19]=C:\Windows\syswow64\ole32.dll [...]

-----BEGIN PGP SIGNATURE-----
Version: GnuPG v1

iQIcBAEBAgAGBQJU3goIAAoJEC9FHFwEZzqVRPYP/3hBSvLWkMnYQLsVnjTcz7DN
43J/ZL6fiXtVqPqpsZ/iGsneyZr0bnfrmtmJdYgFIM1oXNdGgAgNKIbsRHIkpTnZ
cJJAynvmff9r64iClOUW2xs0auKpYfhEq1PmqKUkGp2qw94Oxkx+xXyPb0PIU/yO
Yf3rXe1Y/hpWSC1BtMCdZPeFJ4bbqRGLVatwWsPafsPwco3rFhD4n9eSyXB7jXge
HZ7bDx6FRzfZ9JXCqYM0MuhJEhd66P+jq2jUGnW5qebr7IHzPOxjqtzayJ6NuoNj
z/VT5wIZ50PG6sCJp5Rp9rhY9YiDVuqgzYFzka86DKpAffyyDrM1V47bY5yjNw3o
gVPWO3oJvthNEMa7q6W9vWc83ZJWrszalfMkUDc4dvonjO4Q9W3ffKZX9WUAtuBQ
cg8znaf1yuCQqwKqTSjPsOybzEySdyNIGgR3EPGCaBsiWDxTZDAaHi5LuEv0tSTS
vKZEUnb7nOVRToWi2rh0HWEsYg1Z+1MpIh4uwJ12PODa4fuxMQYYxpDNKphMr73c
hl8+9JESFmniyo00EXbEyqFYN/K++0lM3SIT/IdPyYWzqeGzVLpQv/OvAuqfkgZx
QQZASTizxMjbTfV3gTCZNMXHjIS+Tx6n9sCIV8bnkxGtwcQk8Kzk6DLiYWWdp366
Us6M2dpetDB2y9mLyYnG
=EkWg
-----END PGP SIGNATURE-----

kliment, 2015-02-13:
-----BEGIN PGP SIGNED MESSAGE-----
Hash: SHA1

Unfortunately I've never managed a successful build on a 64 bit
Windows setup. If you figure out how to do it, or why it's not
working, please let me know. I do my builds on a 32 bit setup. The
missing modules should be no problem, everything looks about the same
as on mine. Just out of curiosity, does pronsole work correctly?

> Hi all
>
> I wish to build printrun from source, I've read the #614 issue
> wich explain how to do, but I got some problems anyway.
>
> I'm using Win7 x64 with python27, and Running pronterface.py works [...]
LoadedModule[15]=C:\Windows\system32\IMM32.DLL
> LoadedModule[16]=C:\Windows\syswow64\MSCTF.dll
> LoadedModule[17]=C:\Windows\syswow64\SHELL32.dll
> LoadedModule[18]=C:\Windows\syswow64\SHLWAPI.dll
> LoadedModule[19]=C:\Windows\syswow64\ole32.dll [...]

-----BEGIN PGP SIGNATURE-----
Version: GnuPG v1

iQIcBAEBAgAGBQJU3goSAAoJEC9FHFwEZzqVn9wQAJ40Jb9j3UqhSTbLsDy816Sw
xTA6jlTrtFArTjOsnDC9Py+NxUip51F8GGBgNcfi4KpPFA1HzNpRgpISB+u8gI77
aI1lQe8zIkI6U1vSZK2UvDoLpZsLMGfuzfyD8190IHeA4djLamDtHj/xU/Hp74Fe
oXd/sKUjFoxnoMT5v1YerEWXgWPulMIxYEsOJ7v21YJRTgy//AjyDeHyxDReRtjA
GJmGPsrDYT0+yBc/vmO/5CH+4kgEBUdAwQoDRrJezUIYXbP71MnYMDPVPws6pS/n
TWg6ZPaLH2N6cIzd1KtMjMl71OTCPd954S5KE9mSJD8WH7/r8FE4gvqmD08txuhR
gyzW6ctvQLBgELqkCzKf7l6W5qbKuAQUMKDb0tHXWlJrpFYCFFgNCPmiFZyC+ncy
drHBIi8NIXfX48ubsQU5WNhb6jrIKAU1OZvvVD1odLpndeZZHh+tgBTbljRcRwr9
57nKsuYZ6xoQ2kFbxQb6x+USxMG7Odoi9OS5QKHbpkB3qMa3k76H+JjHrdc2FK7+
NG9aS+PxZqP6KWEQIHxfWEtB7fNwsmVd+2a1xSBnxjADHkTnTH/4tfax0abIpbUt
6PbvkVnoPUm1KCEYGZ92zq3n07z2uEKHUIQ49Ql0Xy4M6o8XkBII98zp71UTLd6V
2X1D9+oSXzpQDOVbFY5E
=k3/f
-----END PGP SIGNATURE-----

gege2b, 2015-02-13:
damn !
I'll try to digg more deeply, but I don't make any promise :)

pronsole didn't run too, he says : 
    WARNING:root:psutil unavailable, could not import power utils:MemoryLoadLibrary failed loading _psutil_windows.pyd

And now that I see this error, I remember that the last time I tried  to run pronterface.exe, a pronterface.log file was created with the same complain, but this time this log wasn't created

kliment, 2015-02-13:
-----BEGIN PGP SIGNED MESSAGE-----
Hash: SHA1

Warning shouldn't make it crash though. At least it's running some
python code - so it's probably one of the libraries causing issues.

-----BEGIN PGP SIGNATURE-----
Version: GnuPG v1

iQIcBAEBAgAGBQJU3hGzAAoJEC9FHFwEZzqVEucP/R9qS6e5xcyTDjeSWOM4EB12
L1CHVvuOz1y8TorAnCeB3dFdcCrN+EVkGQSTE23/mLRKG9QnlUKkZF8bkxzGGjJ/
0+LcVGhqCg2Q+qm6rpF9e9ekKiVF7LJltaWyrVM/xiKvx8P+zEPeSNoCddTc9IzC
9WcKOAFBFDVeSiGjtjIwZAYFynkC/3MLKZvqeW6iqjHUdNIRR+6ioDNty3BG8NVR
4hNnXMrMZahJORIMQ2PRzp/ZoheNNriYWjEE+GEl/GuawoDKkTxAQUrz5uENaCtJ
3QSjpeAIeGn1zXb/vEuRRrJvcdlXSQDC1wV6YHFxbu91Cnv5efLy8aJkuVYDXms0
vQILeqCr2Khj7Wlw/9pGuJZ5fP0Ydak56uLCxESEWJS5eOE5/jpeYZ0xZc0ipcHY
3fmJr3Ns1YtHkSc7zdKsbxt5NLJCN8Cojj1CbeR3xk8PrjU2g93oiawxIZ0UVubT
1hJbt5cil+cV24F6nQ3FQJjLMBdMNL5AfkQaPCTg4n+XPVfoDneUYYZP5H5EypSD
UMSeswQm6iMWXLyOHBVcE//iTLlitJGfAtjn+tNl7SGjAWwo6AbLWkqsxqPiI3eX
vTDp3VEwA1cn82VGgdTW+rXEMpOubXITIr4qTg62EF4TE17XukmSTTuhbXyQhJ/o
6Gk/bL7PibihTa8o3jjP
=zjPW
-----END PGP SIGNATURE-----

gege2b, 2015-02-13:
AH AAAAAAAAH !! SPOTTED IT !
Well, I guess.
And you're right, it's a problem with the libraries (or at least with the way they are processed by py2exe)

I just modified the "bundle_files" option to 3 in setup_win.py, and it just *worked*.

I have not tried to run a print yet, but connecting to the printer, homing, loading Gcode and 3D view works fine
I just get an error on launch, but I it's related to the icon type, as mentioned in #614 (but I did'nt see where to change that ?)

it seems that bundle_files 0, 1 or 2 don't work at all.

The problem with 3 is the huge number of files, but at least it works :)

gege2b, 2015-02-13:
anyway, I have the 32 bit version of python

kliment, 2015-02-13:
-----BEGIN PGP SIGNED MESSAGE-----
Hash: SHA1

Yes, but for some reason I can't get a bundled build process running
on 64 bit windows (even with 32 bit python). So looks like unbundling
files makes it work on those systems, and 64 bit systems can run
bundled binaries built on 32 bit systems just fine. So if you find a
way to bundle it and make it work, that's great but it's not critical.

-----BEGIN PGP SIGNATURE-----
Version: GnuPG v1

iQIcBAEBAgAGBQJU3hzLAAoJEC9FHFwEZzqVay4QAJ9e8u/S/o1mIjioGIpfDWh+
vyNtqJS+KAJLjopbYSxXjYvZZNkY8/CeqMWPVnVnpFC+0i3DYKUVccjH4LiqWj9T
v4ovJcvVkgPqwjKsjNnJ/J69oE58Exp9CRBzn8n+Zn4PEFmZh0v7gFCjHQXSWR7f
CTJDz5F63kh0EYfiESlyQBnobnB2pFZYMW6P8rCAFb2PQi30ni8L9WDP2YTl6dHK
bOVurdnB3CKNm7UD9onAqxMq/6n0HjqL7jSzOdhxSS6D3eXy+Qgc4ov2lmqACLJv
fXO/j5nbNbDasmRwL1mB06X8mG7hAk9Y0LhceUGLfWxoV/iPW+4r5U37GVOlFwkY
KoWYgQbp3L/kwPHDOK65rmA8LGUbeS4aUzy6GvQ4mjX4WbCWjIQc9gvjw0I+zEHF
KgyqNh4ZYf2/E7kcGvCC+Yc2ZnqPF/ly8T5M2ZIDzljl9XJNjVOE9fx67aPNW6Ol
5OUnuUsoSFMHCO+A7XgNW2oKfFT/3OtD62Lg9jB0QIxpL4rXR/ie/471aP+kBkEV
HyTZDN/pBT2K52u9e5CcHSlykZEtPnW6X796TKY2ljMj8/cDy0yvWPAsZc/N91ie
NimYTnHI+vv51+7qKbSS3hSMPLuyKsjs/8jFkyV/1wkJRN4rn+74RxONTPnskUHp
SRCTaGBm2wWSuXTI9ZJq
=nz9s
-----END PGP SIGNATURE-----

gege2b, 2015-02-13:
Ok
After some research, it seems that bundle_files <= 1  is not supported on Win64 plateforms so I don't know if there is any *simple* solution

ntoff, 2015-03-14:
I built it just now on windows 7 x64 and it builds fine, runs "ok" but gives me an error about not being able to load an image from the exe because it's not a png, and the model viewer doesn't work properly.

and I get this in the log

>WARNING:root:psutil unavailable, could not import power utils:MemoryLoadLibrary failed loading _psutil_windows.pyd

which is odd, because psutil is installed and seems to work ok (don't get an error) when running pronterface.py using python 2.7.9

[image]

gege2b, 2015-03-15:
For the PNG error, a line should be modified inside ./printrun/pronterface.py as explained in #614 

Did you build it using the setup-win.py in the repo as is ?

ntoff, 2015-03-19:
Here's what I used http://pastebin.com/7yvXacGu

I just changed the name of the icon and added the import py2exe thing and in the batch file removed the path to python (it's in my global windows path)

gege2b, 2015-03-21:
so, you got a better results than me.

If i'm using bundle_files <= 2 it doesn't work at all, I need to sets it to 3
anyway the error for the PNG is located in ./printrun/pronterface.py, you need to change  "wx.BITMAP_TYPE_PNG" occurences by "wx.BITMAP_TYPE_ICO" at line 184

ntoff, 2015-03-22:
Yeah I fixed that and was able to build a semi perfectly functioning version. I say semi because the plater's are all kinds of messed up but they don't work for me even when run from python but I use this: https://github.com/RobotsWar/Plater as my plater anyway if I need to.

No errors,  no PNG warning, or warnings about psutil of the memory efficient gcode reader. I haven't tested it on a computer other than the one it was built on because I don't have a computer that doesn't already have python + pronterface on it.

DolfKlomp, 2016-03-25:
With regards to the bundle_files < 3 not working on windows 64 bit systems.

It does NOT work for me on my 64 bits windows 7 laptop with 32 bit python 2.7.8 installed (not sure of other packages).
It DOES WORK for me on my 64 bits windows 7 PC with just cleanly installed 32 bit python 2.7.11 and making sure that all packages (including c packages (libpng and freetype not sure if they are needed) were 32 bit.

So in conclusion bundle_files = 1 works for me and is not a specific 64 bits issues.

kliment, 2018-03-23:
We've moved to pyinstaller on the 2.x branch and the build setup works on 64 bit windows too. See https://github.com/kliment/Printrun/blob/2.x/buildinstructions.txt

closed 2018-03-23 by kliment