kliment/Printrun #626: .py error

chief-hacker, 2015-02-24:
We love Pronterface and use it daily.

We also run print jobs lasting over 100hrs. On my last job while printing from Pronterface the print stopped with message displaying there was an error in three of the .py files (I believe it was pronsole, printecore and pronterface.py) tripple shame on me for not writing down the actual error message.

Is there a stripped-down super robust version of Pronterface that will drip feed G-code to my Gigabot?

I don't need slicer, visualization of toolpath, 3D graphics ect...

I am running this on a Microcenter TW100 tablet with Windows 8 and 2gb ram so anythin I can strip out of the program to make it more simple and robust will be a great improvement for me.

Thank you!

iXce, 2015-02-24:
You can easily disable visualization through the options (in Viewer tab,
Main visualization option, select None) !
As for the error, it's pretty much impossible to debug without a proper
backtrace :) If it happens again, could you copy/paste it here?

ntoff, 2015-03-20:
tried the console version? runs in a dos window

chief-hacker, 2015-04-08:
I was able to get a screenshot of the error. 86% complete with a 30hr print then Pronterface quit :( Does this screenshot help?
[image]

rockstorm101, 2017-12-20:
Hi @chief-hacker, I know the issue is quite old, but were you able to solve this problem? In case you are still using Printrun, does this keep happening with the latest release?

Thanks

closed 2018-03-02 by rockstorm101