kliment/Printrun #630: [Feature request] Disable one of the Z axis servos

eliemichel, 2015-02-27:
When I want to get the carriage horizontal, I have to disable one of the Z axis servo.

Nowadays, I do it by hard-unpluging it on the RAMPS, but I believe it would be easy to add some button in Pronterface to disable one of the motors.

kliment, 2015-02-27:
-----BEGIN PGP SIGNED MESSAGE-----
Hash: SHA1

Sorry, they're connected in parallel, there's no software control. As
far as the electronics are concerned, they're the same motor. Also,
they're steppers. not servos. Best you can do is use the "disable
motors" button.

-----BEGIN PGP SIGNATURE-----
Version: GnuPG v1

iQIcBAEBAgAGBQJU8LeiAAoJEC9FHFwEZzqVZqUP/1SNNUQnLZap5VRMqC1e28VC
Bkk1DkXMc6nKXWjkS8vGuPVYC/ktEqqfwAvCl8rXzAhXMAYZglknQiDufoxb11m1
EuYz0Dw8z/yi2KZLFjQX88mowPvf1y2U+MvDjlmGzZki/1PsgRrIVi/U472o/NrE
wLH1UX0iB+y3VJOFv95a05cHJg8ur1s4pxXsjYTyVaL9RIL3jRWihaXGS3kLrITd
+j0SA0/mdmmOthYcrt07nJyWu+lTggJKnQZwY/kwvi9TlV65jX/SWDeW7ejPc2Wr
EIRjd8GNIofXMNs3EQxFchsbHsxIcHQFENrQPT6yeVM8j4sBtODA4NM5hFDXQfum
qu91zUAlyXNb9A2ZN7UugeIbZBBVDy78d+QVavOManiyBVWIOACv3EzlPkpMchCO
bZ6QwG+e5TPPjpuLI+0ZfKUeUnJa4Ta35rJThDPQnxAvxvzlmilj4UXRKAIcfCmS
afzD8lxV5Njsikj5IB1cdJrYp+b9KVv64nMwhxu55u7WVMs6SO/1FWQE65JlF/in
ReMKySv4cLkvEGS5so+9VY3EiV+94+NLE8T8ApU4EH8hyxBr5ELzS8dPxdcLWIzD
asd1uN3fiCFqbqHja9V67opZsheBANqa/aKiyv6GuLvLHyFsjH0KhuqezlZubWEk
jCdz0D/BBaDUB25Eco6z
=QPTA
-----END PGP SIGNATURE-----

eliemichel, 2015-02-27:
Oops, of course they are steppers…
And ok, I understand. I would have to modify the firmware actually.

closed 2015-02-27 by eliemichel

kliment, 2015-02-27:
-----BEGIN PGP SIGNED MESSAGE-----
Hash: SHA1

No, even the firmware doesn't control the Z motors individually.
They're literally wired in parallel.

-----BEGIN PGP SIGNATURE-----
Version: GnuPG v1

iQIcBAEBAgAGBQJU8LqAAAoJEC9FHFwEZzqVvBsP/1rphyl9Na5nmJioZV3llvkc
SxtzajZFHqWfyMi9AxTRWS3GnIi0zPhg0V3zVHJ0wUoCLxTEuIi05OFhq2Ax0cs6
/bJsSCidkM79z21YfA7YiBG5GBswmg3CVJb+JNy+Zt1Bsgz9u9cwqbgVklk9XnYD
ICLRZ+XeOU30SMgniDL3b6Z8LbfOx0KirERbB+2pQxns57psC31WnyK60V6VpGqH
lm8BAsK2UuL52PlbBUfolVO+mJxMVnoHUPfi23kBz+RulpfbxhTj/JjbOpJ+kEf1
PxgWh+e12TTzIcpm8DJNf9X8ncDT/vsYY2k3NkR+HM4zrq3jbkQxciJdBKanCVB6
4hwmFiQ59auOCbequbU+C9EPPtn9kSg3Jhyp6UlpTMptlVnPNMjFlZZmQnJwZDqR
HMVklG5Lt3WsvevNayZz7MYrm8XoNzoLC3+JFJVwH8oUxFmWmRK9Np+n2sAVuHZB
MLD6hhODT4EbtjvvJG6ZMROOBXVVNhQmfuyHeZBKiFxJaqKLrXZVN5Xwbqv5IOU3
hMIc/UQMZd03QH8kQuL+O7BQsY1DM7HQEyDLHF+ncuOntpLcg+Gf2NBh1cEMdECy
q1KBGA4sxLzUvU58CXZnSAYCZafAucYvr+OHzo/48ojqpvR4z02WDsYSXeW5Lpmc
083EOwZBKkx5W5EnRx3n
=nZGt
-----END PGP SIGNATURE-----

iXce, 2015-02-27:
Well, that's not true for all electronics or all setups, you can have 2
separate Z drivers. But anyway, disabling motors is much easier anyway for
leveling!