kliment/Printrun #631: Pronterface <--->  Marlin communication

lvx0, 2015-03-01:
Let me start out with saying that I'm sorry to pull this up is an issue, but all other failed for me.

I'm trying some modifications on the Marlin firmware. 
I pretty much hacked everything. I'm trying to setup a servo based printer, instead of a stepper based.

Now that I'm close to finishing the Marlin code, I'm running into communication issues with pronterface.

I'm not sure what kind of issue it is, and likely some of them are bugs on my side.
So my question is, what is it that pronterface needs to see coming back from Marlin?
Is there some kind of communication protocol/manual you guys agreed on.

best regards
LVX

kliment, 2015-03-01:
-----BEGIN PGP SIGNED MESSAGE-----
Hash: SHA1

> Let me start out with saying that I'm sorry to pull this up is an
> issue, but all other failed for me.
>
> I'm trying some modifications on the Marlin firmware. I pretty much
> hacked everything. I'm [...]
Uhm, yes, there is a communication protocol. You need to acknowledge
the receipt of every line, check line numbers and checksums. You can
look at printcore.py to see how the actual communications are
implemented on the pronterface side.

> 
> best regards LVX
> 
> https://github.com/kliment/Printrun/issues/631

-----BEGIN PGP SIGNATURE-----
Version: GnuPG v1

iQIcBAEBAgAGBQJU842TAAoJEC9FHFwEZzqVFiQP/34gHHG0u6ghP7nkw2KBn4zT
1L4Pxs4uuzeonbONO2pauwwg+8+dvpWMAUXTZrj7q9rAbt2Kgnu8J35NZd5Jxnh1
SmptmZVFIlO4lCcHvLv3zxEUO8AhRzLtRWwYFIp1K6uOO+XlKyDxcQd2Lpf2bFpK
y0XaPvvMB0ydjYV5FzpcY/7LMr3GcfwZFAwgyz94rLV4C/v2gMSwCxYwews+4CUb
djOykgXoxMOt7YTFffF8CYch6iEoG760DqfQY5d7MyyEohuLz5OuzBH0o0jd4HAX
mU44EFRcQh1jotJmREZbjEVEE6sam6Vd3+b/3lKW+cGm37LVOT6UbL6bngL+9PU/
AgboE0yN6j7NBxmUpALu3JM+6VtQGUf4OyX071SM4f2I+jYwgUtFAMk4qon+Atm5
vtmrYiwrQySPXUpOwY4j6Ex/J9YVP4BKVCzUQUSSjO0V6Btf7mM4lo6/gFeVTP1t
IxRs8CUUrBbJb1R2WI+LuMwkoPKuq7UlOcEQMIcC5T/nbPAFVQdTbFni7lxjIBx/
sATTIf5xYdmjjy5CeqO15yMwf5p4CxMqc0XkQMT9U3TXqw009FEMjZmmh13+0c1g
0x+Cxypls3ud9ZrfxurfhmyuFJCpWbkKm48pZ9ca+roMK2zOK/VcrB3o6lMaia9x
/RH2bOaNsxzBsu9YnHo6
=yrsQ
-----END PGP SIGNATURE-----

iXce, 2015-03-01:
Isn't the communication protocol the standard CNC/gcode one ?

kliment, 2015-03-01:
-----BEGIN PGP SIGNED MESSAGE-----
Hash: SHA1

Standard for repraps, but many cnc setups use different arrangements

-----BEGIN PGP SIGNATURE-----
Version: GnuPG v1

iQIcBAEBAgAGBQJU849LAAoJEC9FHFwEZzqVO0wP/0vuUYSZBYAczggrmNNK2zqy
H4vsyrUUjadjztg0dlqm/XNUILZwWWaSK2MPKa1eBxXkwvR6B5uRxzw5aiiKrTmt
bncH4i4au7BTT19oberSAo1GApXy4w++iTRGsal4TDvmMsApvaag9WNltf3WK4or
pplx5K+qa6YiOssgIvRmB7EuPRCrOoOCZBz3Ggpp+F5cxFBkGf0UppWbdj42EbSu
EuAfGGs9wjk8hn7EwbEqmJVxYxMVhdlmdGSAp55Kuv/Q9bqp6uusnkY7uFSbPPUp
ryCxRxpHBhM2+6zNecOoGPvDcNlg+PZYjtcU4hzQm4ZLr7Z2fgIL6r90HnMXDTOW
RiS7DvfwZogri6CywCVl5GRakPAfcV0SxlZ7Ln8l4CtA6avyr4M7H8fSlPrx1vPB
QxiSeXDKoNz965kpzg3xH9cM1Irm4ICGL/m+M9wYb6b5GumIu0EMtd0+GVeoGYCy
3u0/QX73PG3H4NO5VpJvMb7K7Db9MspuBiZSVEGaXrcf29303YXthu8wJ+9dcILt
ZbteJsMReHz/ytjOn0pVD0WCL8CDuZN24ooWBtH1H6A3rmW1NmYWZVkMPZp2YI5u
Ci0Ol+rVRyTKMrvSydEzf/5awCsCUHGluerW6KwHFcsmUpP8Ti9nZ33tBbgUzFMY
7EyWSXUgjPjn9siD/osY
=MrDR
-----END PGP SIGNATURE-----

JustAnother1, 2015-03-02:
The protocol is specified here:
http://reprap.org/wiki/Gcode

lvx0, 2015-03-03:
Thank you all for your replies. That clarifies a lot.

I'm still experiencing some trouble, but at the moment I can't detect if the issue is in the software or in the FTDI chip on the Arduino Mega 1280.

Most of the commands coming from pronterface arrive correctly, but someone are just not read completely. Thus giving me checksum errors.

closed 2015-03-03 by lvx0