kliment/Printrun #632: Pronsole disconnects after each command, when run externally

hornborg, 2015-03-02:
Hi,

Trying to run pronsole from another python program.

Example of my code:
proc = subprocess.Popen('D:\Program\Printrun\pronsole.exe',stdin = subprocess.PIPE)
proc.communicate('help')

Gives me:
"[ERROR] Not connected to printer.
Welcome to the printer console! Type "help" for a list of available commands.
uninitialized> 
Documented commands (type help <topic>):
========================================
bedtemp     exit     help  macro    pause   reverse  shell 
connect     extrude  home  monitor  print   sdprint  skein 
disconnect  gcodes   load  move     reset   set      tool  
eta         gettemp  ls    off      resume  settemp  upload

Undocumented commands:
======================
extrude_final

uninitialized> 
Disconnecting from printer...  
Exiting program. Goodbye!"

As you can see it's accepting my command 'help', but it disconnects directly before I can give any more commands. Any ideas?
Thank you in advance!

kliment, 2015-03-02:
-----BEGIN PGP SIGNED MESSAGE-----
Hash: SHA1

It's encountering the end of input and terminating. Why don't you use
pronsole.py directly instead and just call its functions?

-----BEGIN PGP SIGNATURE-----
Version: GnuPG v1

iQIcBAEBAgAGBQJU9NiPAAoJEC9FHFwEZzqVuRsP/3wYOCtlSL/HqRoy7TS7YPCT
ZWWGVrf7/IdlwqYm5APGeOqxgUoQeHIqyyLaNGJizdENDHmR/7WS05Klg6BxiZ34
Jsj/G+93CWGVNQSDScTQ8uStac/PN3PkdG6sg8YMiuQGQzqyB2KAiphL8TIaNLvJ
LVyDQwTa+v6zXxwQXMIEBJoLY/L7S/f43RwAwKwoYuPjRW4K/3OOEmDipWy8/4hO
E1mGUh/jOzSOvVpIhmkd90XghLfHHXkLJ08ra4YT1Mjsgd6d0tNwzF76o2LT2ws9
SfipoSxsn82Xk+QnkE+BVbIabpu20HQdRTfZ0vIVk9SAs1NkKD9wen+KkybrM11+
BlCd1GCOXQD9C6WQnvPHk4V0goM0dzKfXD3fvVuyyfBryRHbPPOzDNVyTiNPbCio
dFXOl5OPHABaitnn+b62dI3Q4zHmA7d6ZiNZHhydxotKWU7/Txz8ppizffYq9tLE
elMikldV7fYD45+Z+2QadQehHPCN0dyTUvlPxP7VBdQGYOyduK+Aq1J99t9sxQIL
O255AJDklRiy51yMgpP/R8Kl+4FAZy0XJszr3R1fFLDOAoT6P8YK7cqY2ZmD8TtS
mH+XHS8k4fuAgAUEMT8ywSqxgQi3riRueFZzgDxaPZ2+IY7cdksbMCwNPg5vhnVm
VEEtnA8+akx1iE0TXPiK
=wo2w
-----END PGP SIGNATURE-----

iXce, 2015-03-02:
Isn't it because of the specific "communicate" function which puts an EOF ?

rockstorm101, 2017-12-19:
Hi @hornborg, was your question solved? If so, could you please close this issue?

closed 2017-12-19 by hornborg