kliment/Printrun #634: Simple request: Get estimate Finish Time

JFPayeur, 2015-03-09:
Could it be possible to add the estimate remaining time to the host computer time. I'm shure people may care more to know when the print will finish than in how much time and add it up in there head. 

I know I'm lazy :P I'm shure it's just a dozen of line of code ( or less :) )

ntoff, 2015-03-14:
Isn't that what it does? I'm sure it says estimated time x hours x minutes left of <total print time>

kliment, 2015-03-14:
-----BEGIN PGP SIGNED MESSAGE-----
Hash: SHA1

I think the request is to display the time when it would finish, in
addition to how long there is until then. So if local time is 2 pm and
it will take one more hour, it would display "Finishes in 1 hour, at
15:00". At least that's how I understand it.

-----BEGIN PGP SIGNATURE-----
Version: GnuPG v1

iQIcBAEBAgAGBQJVBJgrAAoJEC9FHFwEZzqVB74P/RMO2AVYn2QO7EPL7Ycstpkx
RvxvcFAbl2qj6Pu9lIQM5HSMWygR5YnrhrEHmtNEsKeoYGUiXeKt6JoP7o7RY+3x
zid03uyusVudyqRwgA4719ePYUBpBFjwHXonceCPbSTtMVB0KBsKcoAx9qNatMtU
gvsh5mnmXURFqEUW4Ik4UhHGudqPHLOV/MCnjrFnrJNj4aChoQJca3yt2dlgQqUM
0NZHQXE6DU2OGS1c385hg8EUNEgwwIp3mtH0hVVFPMAgId9ODAlZyy2zFXpTf+v+
dNAMMc2MitKqzT5JHot7qnfyq5gdJpSesvERuVNEBys+CQiGMmQUky9dEXH5oLIA
yxuyr3CWnKb6KxV6OBERLrSCfu+YZQw1NkN49umJWkVZDjh/PRLxPDKwkuMVYXGs
R5mMu7tzCwaaN2nLkSLxM1HOzgPrm15qbmKS98c4NzxTiQF43rgTkd2afLtB9Jh1
kax+wSdlCPoSX8kmpk0Ucbk2AwScZlQl5T2AZY+SZlcKKAD4D73ClFk4uoY0MIEZ
ZjBUQHNWrlgVE7q+G9ZdH4PD27Dqyi7NCY2TfWPnA1SjVJZHOaeHlsRK8WOlrLRr
nQdMkZlqgOLmD8OwbQMAGiwFj3ur/bKPY/rPJnbvYPewkc3+K7FJefs1y8tBdlPo
XYKuPITCsAw5jITMvugp
=56jr
-----END PGP SIGNATURE-----

ntoff, 2015-03-19:
to build on this, I had a look earlier but I know nothing of python, is it possible to make it send "M117 (estimated time here)" to the printer say every minute? That'll display the ETA to finish on the LCD screen (if the printer has one).

iXce, 2015-03-19:
Hmm, about the initial request, I think it may take just too much space, but that's something we could add an option for (display ETA or remaining time).
About that M117 request, it's an excellent idea. Do we know how wide is M117 support ?

ntoff, 2015-03-20:
wide? the number of characters? It depends on the LCD panel used. 16x4 and the reprap full graphics display I think are the most common. 

http://reprap.org/wiki/RepRapDiscount_Full_Graphic_Smart_Controller

Repetier host has this as a feature and I think it just prints out "ETF: xh:ym:zs" which will easily fit on a 16x4 LCD display.  I would keep it as low as possible though, once I sent a really long string while it was printing and the printer kind of spazzed out.

You'd want to limit it to updating maybe once every 30 seconds or so as well so you don't overload the buffer with nothing but M117 commands and leaving no room for the actual printing. I tried it last night (1am so not much time to test) and it flooded the printer and froze the Arduino.

iXce, 2015-03-20:
No I meant for the initial bug report, to add the finish time in the status bar !

ntoff, 2015-03-20:
Ok I got it sort of working and it seems to update on layer change. I'm not a coder so I don't know how reliable it is. Also it doesn't do days. If the time is over 24h, it'll just show that as hours. e.g. 45h 17m 32s

ntoff, 2015-03-26:
Well, looks like the end time thing works, so does the LCD ETA. Although the LCD only displays the time left, not the actual end time because for some reason sending a : (colon) through M117 ends the transmission and I don't know how to get around that.

I have no idea how this whole github pull request thing works though but I have it up on my github if anyone wants to go test it.

Beware: I'm not a coder, I have no idea what I'm doing. Don't go using it in a production environment or on a critical print. It's just there so someone who does know what they're doing can take a look at it and see if it's worth looking into.