kliment/Printrun #635: Which packages are needed for applications?

pOmelchenko, 2015-03-09:
Hi!

Yesterday I installed Ubuntu 14.04.2, just installed a set of applications and drivers to work. Now I set Pronterface and I can not figure out the list of packages from the readme.

I downloaded the app from here and the list of the readme established only those packets which were obvious mistakes.

After cloning required to establish the following:

python-wxgtk2.8
python-pyglet
python-psutil
python-numpy
cython
python-cairosvg

But were still in the list, and these packets:

python-tornado
python-setuptools
libavahi-compat-libdnssd1

When the program error I did not see, on this question, and they all need? I do not interfere with them, too, to establish just want to understand their necessity.

Here are the logs of the APT in order to enable you to analyze my actions and installing updates, if necessary, update the list in the readme file:

https://drive.google.com/open?id=0B9x_cXM_rmjdemNNVWRIaEJpaWs&authuser=0
https://drive.google.com/open?id=0B9x_cXM_rmjda2lEeGFYUjY1MDg&authuser=0

iXce, 2015-03-09:
Hm, I'm not sure what's your issue ? I understand that some packages may not be necessary anymore compared to the README, and I'm going to update that, but is that the only issue or do you have trouble launching printrun ?

pOmelchenko, 2015-03-09:
No problems with the launch and operation of the application no. Just a question of the need for some packages that are listed in the readme file.

If I understand you correctly, packages python-tornado, python-setuptools, libavahi-compat-libdnssd1 no longer used in the applications?

iXce, 2015-03-09:
tornado and avahi stuff are definitely not needed anymore, not 100% sure for setuptools, but anyway it would not be a runtime dep but a buildtime dep to make source packages, so not something that you have to worry about !

pOmelchenko, 2015-03-09:
Cool! Thank U!

closed 2015-03-09 by pOmelchenko