kliment/Printrun #637: impossible to install

matphillips, 2015-03-15:
I've tried the precompiled binaries on OS X and windows, neither work.
I've followed the instructions for installing from source for both too, the windows ones produce nothing but errors and the OS X ones make no sense in the first place.

Is there any chance of a working binary, or usable instructions?
And before anyone throws a "works for me" response, please try to install it on a machine that's never seen it before, I assure you, it won't work.

Sample of the many errors seen: 
[ERROR] Failed to execute slicing software: 
Traceback (most recent call last):
  File "/Users/craigvanvliet/Desktop/PyInstaller-2.1/pronterface/build/pronterface/out00-PYZ.pyz/printrun.pronterface", line 1257, in slice_func

I don't know who craigvanvliet is, but I guess he's wrk compiled the OS X binary, so I guess it does in fact, work for him.

iXce, 2015-03-15:
Hmm, how does it "not work" ? What are the errors ? The code you mention is triggered when you slice a stl, so when loading a file, so I imagine you can at least start the program and it kind of works ?

matphillips, 2015-03-15:
Indeed. It launches on OS X and shows that error and more when trying to open an STL. The windows version does similar but I'm not at that machine right now to check.
It's unclear exactly how to launch the windows one from downloading source, but I've tried opening pretty much all the python scripts and can't get a GUI at all.

I do realise that this isn't much to go on, but after trying three machines (two laptops and a desktop at work) I'm pretty sure the symptoms I'm seeing are replicable on other computers.
Thanks.

iXce, 2015-03-15:
Does loading a gcode file works ? I imagine that there's a problem in the slicer stuff (either the slicer path is wrong or the slicer won't produce gcode or something), but that the rest may work as expected !

iXce, 2015-03-15:
(to check that the slicer is where it is expected to be, you can go to Settings => Slicing settings and check that Slic3r's window appears.

matphillips, 2015-03-15:
Ok, so Slicer turned out not to be installed on this laptop so I resolved that, then opening an STL (a set of chess pieces as a sample file downloaded from thingiverse (61774) produces this:
Slicing /Users/phillipsm/Dropbox/3dmodels/chesswireframe.stl
Running /Applications/Slic3r.app/Contents/MacOS/slic3r /Users/phillipsm/Dropbox/3dmodels/chesswireframe.stl --output /Users/phillipsm/Dropbox/3dmodels/chesswireframe_export.gcode
=> Processing triangulated mesh
No layers were detected. You might want to repair your STL file(s) or check their size and retry.

Loading file: /Users/phillipsm/Dropbox/3dmodels/chesswireframe_export.gcode
Exception in thread Thread-5:
Traceback (most recent call last):
  File "/Users/craigvanvliet/Desktop/PyInstaller-2.1/pronterface/build/pronterface/out00-PYZ.pyz/threading", line 552, in __bootstrap_inner
  File "/Users/craigvanvliet/Desktop/PyInstaller-2.1/pronterface/build/pronterface/out00-PYZ.pyz/threading", line 505, in run
  File "/Users/craigvanvliet/Desktop/PyInstaller-2.1/pronterface/build/pronterface/out00-PYZ.pyz/printrun.pronterface", line 1372, in load_gcode_async_thread
  File "/Users/craigvanvliet/Desktop/PyInstaller-2.1/pronterface/build/pronterface/out00-PYZ.pyz/printrun.pronsole", line 886, in load_gcode
IOError: [Errno 2] No such file or directory: u'/Users/phillipsm/Dropbox/3dmodels/chesswireframe_export.gcode'

That same file opens right away in Cura.

So I restarted and tried a different file and that seems to just crash the app at "Processing triangulated mesh" giving just the spinning beachball and leading to the app needing to be force quit.

iXce, 2015-03-15:
Mind uploading that 3d models ? For the second case, can you check if slicer is actually running and doing stuff ?

matphillips, 2015-03-15:
http://www.thingiverse.com/download:154301

For the second one, the slicer icon is bouncing in the dock, but there's no window for it so I can't say if it's doing anything or not.
I tried again, applying more patience this time and noticed that it does generate a gcode file in the same location as the STL, and eventually the slicer icon went away but printrun stays unresponsive.
Thanks.

iXce, 2015-03-15:
The object in that file is super tiny (print-dimension wise), it has to be scaled up at least 100 times I think, this is probably why Slic3r does not detect any layer.

As for the second problem, if Slic3r bounces I guess it means slicing has succeeded, can you check if a gcode file has been produced in the same folder as the original stl ?

rockstorm101, 2017-12-19:
Hi, @matphillips, I know the issue is old but, what the status of if? Were you able to get Printrun/Slic3r to function properly?

matphillips, 2017-12-19:
I don't think I ever did, but I now use the builds provided by Prusa and they work just fine.

closed 2017-12-19 by matphillips