kliment/Printrun #641: famous gcoder_line error, wrong paths in setup.py

indazoo, 2015-03-18:
Hi
I would like to start printrun without the annoying gcoder_line error message at each start and of course use less memory.
Message:
WARNING:root:Memory-efficient GCoder implementation unavailable: No module named gcoder_line
*************************************************************
I work on Windows 7
I installed all the follwing on my system to get Printrun up ( a single program!!! ).
- python 2.7.9
- pyserial
- wx python
- pyreadline
- pyglet
- numpy
- MinGW + compile Cython ( I thought I need that to compile Cython but it created just a ton of compile warnings)
- pyCairo
- CairoSVG
- Cython from http://www.lfd.uci.edu/~gohlke/pythonlibs/#cython ==> install worked without errors
- later p2exe
*************************************************************
Then i run "python setup.py build_ext --inplace" in printrun directory

> WARNING:root:Memory-efficient GCoder implementation unavailable: No module named
 gcoder_line
> running build_ext
==> nothing else
*************************************************************
Possible bug:
Command "python.exe setup.py install"  creates two errors:
1. ValueError: path 'share/pronterface/images/' cannot end with '/'
2. ValueError: path 'share/pixmaps/' cannot end with '/'

Because of setup.py:
target_images_path = "share/pronterface/images/"
data_files = [('share/pixmaps/', ['pronterface.png', 'plater.png', 'pronsole.png']),
              ('share/applications', ['pronterface.desktop', 'pronsole.desktop', 'plater.desktop']),
              ('share/appdata', ['pronterface.appdata.xml', 'pronsole.appdata.xml', 'plater.appdata.xml'])]

*************************************************************
After fixing the bugs(?)  the command 
 >   python.exe setup.py install  
works

the command 
python setup.py build_ext --inplace
works also, but the message about the gcoder_line s still present

Any help appreciated

iXce, 2015-03-18:
Hmm, does the python setup.py build_ext --inplace produces something ? Do you have a list of files called gcoder_line.* in the Printrun file tree ?

indazoo, 2015-03-18:
Sorry for the late response, had the kids to watch...
Here is a list of files with "gcoder" in name
C:\Users\User\AppData\Roaming\PrintRun_20150310\build\lib\printrun\gcoder.py
C:\Users\User\AppData\Roaming\PrintRun_20150310\printrun\gcoder.py
C:\Users\User\AppData\Roaming\PrintRun_20150310\printrun\gcoder.pyc
C:\Users\User\AppData\Roaming\PrintRun_20150310\printrun\gcoder_heapy_support.patch
C:\Users\User\AppData\Roaming\PrintRun_20150310\printrun\gcoder_line.pyx
C:\Users\User\AppData\Roaming\PrintRun_20150310\printrun\gcoder_line_extra.h

indazoo, 2015-03-18:
Command:
C:\Users\User\AppData\Roaming\PrintRun_20150310>python setup.py build_ext --inplace

Output (2 lines):
WARNING:root:Memory-efficient GCoder implementation unavailable: No module named gcoder_line
running build_ext

iXce, 2015-03-18:
Okai so its not even trying to compile it I guess. Gotta find a Windows
setup

Command:

indazoo, 2015-03-19:
Hmm, i did some research...

Setup.py fails at:
    extensions = cythonize("printrun/gcoder_line.pyx")

I inserted a print statement which gives:
"DLL load failed: Invalid access to memory location."

Oh Oh ....
http://bugs.python.org/issue16472

I admit at the moment I have not understand it fully....can you help?

PS.:
I added an issue here on github about the print statement in the exception

indazoo, 2015-03-19:
I tried installing the MS compiler

Microsoft Visual C++ Compiler for Python 2.7 
http://www.microsoft.com/en-us/download/confirmation.aspx?id=44266

and uncommented mingw32 in distutils.cfg

Same "dll load failed" message

Then i created a hello world sample from here:
http://en.wikibooks.org/wiki/Python_Programming/Extending_with_C
This compiled correctly to the pyd file

indazoo, 2015-03-19:
Both compilers seem to work

MinGW command:
python setup.py build --compiler=mingw32          (and with mingw32 in distutils.cfg)

Result:
running build
running build_ext
building 'hello' extension
creating build
creating build\temp.win32-2.7
creating build\temp.win32-2.7\Release
C:\Program Files (x86)\Haskell Platform\2013.2.0.0\mingw\bin\gcc.exe -mdll -O -W
all -IC:\Python27\include -IC:\Python27\PC -c hellomodule.c -o build\temp.win32-
2.7\Release\hellomodule.o
writing build\temp.win32-2.7\Release\hello.def
creating build\lib.win32-2.7
C:\Program Files (x86)\Haskell Platform\2013.2.0.0\mingw\bin\dllwrap.exe -mdll -
static --entry _DllMain@12 --output-lib build\temp.win32-2.7\Release\libhello.a
--def build\temp.win32-2.7\Release\hello.def -s build\temp.win32-2.7\Release\hel
lomodule.o -LC:\Python27\libs -LC:\Python27\PCbuild -lpython27 -o build\lib.win3
2-2.7\hello.pyd

VC compiler command:
python setup.py -v build

Result:
running build
running build_ext
Importing new compiler from distutils.msvc9compiler
building 'hello' extension
Calling 'vcvarsall.bat x86' (version=9.0)
creating build
creating build\temp.win32-2.7
creating build\temp.win32-2.7\Release
C:\Users\User\AppData\Local\Programs\Common\Microsoft\Visual C++ for Python\9.0\
VC\Bin\cl.exe /c /nologo /Ox /MD /W3 /GS- /DNDEBUG -IC:\Python27\include -IC:\Py
thon27\PC /Tchellomodule.c /Fobuild\temp.win32-2.7\Release\hellomodule.obj
hellomodule.c
creating build\lib.win32-2.7
C:\Users\User\AppData\Local\Programs\Common\Microsoft\Visual C++ for Python\9.0\
VC\Bin\link.exe /DLL /nologo /INCREMENTAL:NO /LIBPATH:C:\Python27\libs /LIBPATH:
C:\Python27\PCbuild /EXPORT:inithello build\temp.win32-2.7\Release\hellomodule.o
bj /OUT:build\lib.win32-2.7\hello.pyd /IMPLIB:build\temp.win32-2.7\Release\hello
.lib /MANIFESTFILE:build\temp.win32-2.7\Release\hello.pyd.manifest
   Creating library build\temp.win32-2.7\Release\hello.lib and object build\temp
.win32-2.7\Release\hello.exp

indazoo, 2015-03-19:
Hi I  tracked it down.
In the process of installing all those packages (pip, msi, manual installs) I had the following Cython on my disk:
C:\Python27\lib\site-packages\cython-0.22-py2.7-win32.egg\Cython

With a even more verbose output  I could see more (see my comment in the closed 642 issue, would be nice if you can add that too) 

The command:
"python setup.py build_ext --inplace --compiler mingw32"

showed the following stack trace:
C:\Temp\PrintRun\Printrun-printrun-20150310_test>python setup.py build_ext --inp
lace --compiler mingw32
Exception while importing modules:
DLL load failed: Invalid access to memory location.
Traceback (most recent call last):
  File "setup.py", line 28, in <module>
    extensions = cythonize("printrun/gcoder_line.pyx")
  File "C:\Python27\lib\site-packages\cython-0.22-py2.7-win32.egg\Cython\Build\D
ependencies.py", line 745, in cythonize
    ctx = c_options.create_context()
  File "C:\Python27\lib\site-packages\cython-0.22-py2.7-win32.egg\Cython\Compile
r\Main.py", line 537, in create_context
    self.cplus, self.language_level, options=self)
  File "C:\Python27\lib\site-packages\cython-0.22-py2.7-win32.egg\Cython\Compile
r\Main.py", line 69, in __init__
    from . import Builtin, CythonScope
  File "C:\Python27\lib\site-packages\cython-0.22-py2.7-win32.egg\Cython\Compile
r\CythonScope.py", line 5, in <module>
    from .UtilityCode import CythonUtilityCode
  File "C:\Python27\lib\site-packages\cython-0.22-py2.7-win32.egg\Cython\Compile
r\UtilityCode.py", line 3, in <module>
    from .TreeFragment import parse_from_strings, StringParseContext
  File "C:\Python27\lib\site-packages\cython-0.22-py2.7-win32.egg\Cython\Compile
r\TreeFragment.py", line 17, in <module>
    from .Visitor import VisitorTransform
  File "Cython\Compiler\Visitor.py", line 14, in init Cython.Compiler.Visitor (C
ython\Compiler\Visitor.c:17356)
  File "C:\Python27\lib\site-packages\cython-0.22-py2.7-win32.egg\Cython\Compile
r\ExprNodes.py", line 3846, in <module>
    class SliceIndexNode(ExprNode):
  File "C:\Python27\lib\site-packages\cython-0.22-py2.7-win32.egg\Cython\Compile
r\ExprNodes.py", line 3998, in SliceIndexNode
    "SliceObject", "ObjectHandling.c", context={'access': 'Get'})
  File "Cython\Compiler\Code.py", line 277, in Cython.Compiler.Code.UtilityCodeB
ase.load (Cython\Compiler\Code.c:7968)
  File "Cython\Compiler\Code.py", line 464, in Cython.Compiler.Code.TempitaUtili
tyCode.__init__ (Cython\Compiler\Code.c:12430)
  File "Cython\Compiler\Code.py", line 457, in Cython.Compiler.Code.sub_tempita
(Cython\Compiler\Code.c:12155)
  File "C:\Python27\lib\site-packages\cython-0.22-py2.7-win32.egg\Cython\Tempita
\__init__.py", line 4, in <module>
    from _tempita import *
ImportError: DLL load failed: Invalid access to memory location.

WARNING:root:Memory-efficient GCoder implementation unavailable: No module named
 gcoder_line
running build_ext

Very ugly hack:
I copied Tempita folder from the other Cython install "C:\Python27\lib\Cython" to the "egg-Cython" folder because the two pyd files were really different and the dll load error no longer ocurred and printrun compiled and started up.
==> So this "egg-Cython" has the wrong VC version linked

I made another to-be-secure test:
Cython with "pip install Cython-0.22-cp27-none-win32.whl" has the wrong "Tempita" version
Cython with "easy_install cython" come with the correct "Tempita" version.

All good now..case closed for me. I'll give notice to gohlke. Thanks for your feedback.

P.S.
To be precise, first I tried to compile with the Windows Compiler but it did not found the file "stdint.h".
It is not in "C:\Python27\include" or maybe missing somewhere else
...well...

indazoo, 2015-03-19:
Here is the "sucessfull" MinGW compiler output. Perhaps one has something to do:

C:\Temp\PrintRun\Printrun-printrun-20150310_test>python setup.py build_ext --inp
lace --compiler mingw32
Compiling printrun/gcoder_line.pyx because it changed.
Cythonizing printrun/gcoder_line.pyx
WARNING:root:Memory-efficient GCoder implementation unavailable: No module named
 gcoder_line
running build_ext
building 'printrun.gcoder_line' extension
creating build
creating build\temp.win32-2.7
creating build\temp.win32-2.7\Release
creating build\temp.win32-2.7\Release\printrun
C:\Program Files (x86)\Haskell Platform\2013.2.0.0\mingw\bin\gcc.exe -mdll -O -W
all -IC:\Python27\include -IC:\Python27\PC -c printrun/gcoder_line.c -o build\te
mp.win32-2.7\Release\printrun\gcoder_line.o
printrun/gcoder_line.c: In function '__Pyx_RaiseArgtupleInvalid':
printrun/gcoder_line.c:5297:18: warning: unknown conversion type character 'z' i
n format
printrun/gcoder_line.c:5297:18: warning: format '%.1s' expects type 'char *', bu
t argument 5 has type 'Py_ssize_t'
printrun/gcoder_line.c:5297:18: warning: unknown conversion type character 'z' i
n format
printrun/gcoder_line.c:5297:18: warning: too many arguments for format
writing build\temp.win32-2.7\Release\printrun\gcoder_line.def
C:\Program Files (x86)\Haskell Platform\2013.2.0.0\mingw\bin\dllwrap.exe -mdll -
static --entry _DllMain@12 --output-lib build\temp.win32-2.7\Release\printrun\li
bgcoder_line.a --def build\temp.win32-2.7\Release\printrun\gcoder_line.def -s bu
ild\temp.win32-2.7\Release\printrun\gcoder_line.o -LC:\Python27\libs -LC:\Python
27\PCbuild -lpython27 -o C:\Temp\PrintRun\Printrun-printrun-20150310_test\printr
un\gcoder_line.pyd

C:\Temp\PrintRun\Printrun-printrun-20150310_test>

indazoo, 2015-03-19:
now I'm off...really

closed 2015-03-19 by kliment