kliment/Printrun #642: setup.py silently dies

indazoo, 2015-03-19:
Hi
In issue https://github.com/kliment/Printrun/issues/641 I found after long debugging a supressed error message. it's in setup.py. 
Below is an idea how to inform the user a bit better so the person don't have to search for hours on the internet. 
Damn!!! Ok,  I just needed that. Excuse me.

try:
    from Cython.Build import cythonize
    extensions = cythonize("printrun/gcoder_line.pyx")  <== This failed
     from Cython.Distutils import build_ext
except Exception, e:     <==  changed code
    print "Imports failed"    <== new code
    print str(e)                   <== new code
    extensions = None
    build_ext = None

iXce, 2015-03-19:
Verbosity added =)

closed 2015-03-19 by iXce

indazoo, 2015-03-19:
Thanks a lot.
I'm sorry I solved Issue 641 only thanks to a even more verbose output

try:
    from Cython.Build import cythonize
    extensions = cythonize("printrun/gcoder_line.pyx")
    from Cython.Distutils import build_ext
except Exception, e:
    print "Exception while importing modules:"
    print str(e)
    print traceback.format_exc()
    extensions = None
    build_ext = None

With that I could see where the dll load failure happened. More comments in a few minutes in 641
If you could add the traceback thing .. Thanks

iXce, 2015-03-19:
Well usually people get super afraid of tracebacks, and it's only a
recommended addition, not a mandatory thing, so I'm not sure what's the
best thing to do here :/

kliment, 2015-03-19:
-----BEGIN PGP SIGNED MESSAGE-----
Hash: SHA1

I agree the traceback looks scary and shouldn't be there. Maybe have
it in the code but commented out so that people who are debugging can
enable it.

-----BEGIN PGP SIGNATURE-----
Version: GnuPG v1

iQIcBAEBAgAGBQJVCr4lAAoJEC9FHFwEZzqVLW4QAIe0anviPsjmlW/kI4QPBGyo
BeK+sXU9aupG9MmLy039DP8JDWb+oY0WoTTK4Nz4o0+Z4/KAPsy8OVT7Dg0zPgtG
XJOkADqNv/SLcKEoqHuRZiq9Rav6+ZVWMYmixe+fzaMKglRWdSpuZd2Meanta9+e
9Sqi/4YILCig5PXgceqh0HPTMEren+Ls0agC/QKyHTyAUzUwotenHPxhabJ4Wuwk
O9dFsuPhJDl8Ar+J/Df75Ro4VPL+fb5SjokHUZA6p5+advm///JxWoz45n03tQRZ
Gi+OC6OwFzCAZyj04hPMwgLKLbQDC3+2RasyFRP6wNUjf+DDnokBJGApAIKLziyL
/wqJbSZrLStJLU+TzOpISKI8UzOl3dP1h31bezS3S5QcGQtBTJw/09mmfvfHzkOZ
z1QzoPJkZYVogPbYUoafJt6blgv317+g+sbOuCCb3V7In7GufSfhWPlwsjCu1aLD
KXJnfLU48RhqO3wO39FbYOgxDZC3XTP8hcduRZbu2fj49KkS3HIfnMGaVuQf8mHq
VCTjIqHLm3b9Xr0XuBdspE/N1VQ5w49u5Nln/rb6DoIjMIPDH3sBcWGw3yyktyqp
X3mVPtYKWyyYccMZEumRo5fBD0FDSzE80oJmM6xZq2kPso/Y3greMo5b18fRPLPX
CGJk71ZNhdnjjNTpek5Y
=hSxX
-----END PGP SIGNATURE-----

iXce, 2015-03-19:
Agreed & pushed !