kliment/Printrun #643: Pronterface could not open port (error5)

dadatada, 2015-04-01:
Hello.
I need some advice, this is my first 3d printer.... just installed everything like was in a nexdayreprap booklet-similar to kliment install list. 
But when try to comunicate with the printer it says "connecting" all the time... when I click it once more said "python serialexception could not open port/error 5 access is denied"
What I'm doing wrong?
I tried to reinstall all once more but no help. Tried to change a port, update driver with older 2560...

my components are:
Asus R2H on windows 7 32bit
Funduino mega v3
ramps 1.4 ultimachine
a4988 drivers

Thanks in advance

msoya, 2015-06-24:
I have the same problem. While i am triying to connect from pronterface to arduino by COM port 13, it says "connecting" but nothing happens for minutes. At the end the log file is like that:

"serial.serialutil.SerialException: could not open port COM13: [Error 2] system can not find the specified file.
serial.serialutil.SerialException: could not open port COM13: [Error 5] access denied."

I tried to open the program with administrator mode, nothing change.
I tried to change the COM port number, nothing change.
I tried to disable and able the COM port, nothing change.

My operating system is Windows 7 64 bit.

Thanks your comments already..

aboobed, 2015-06-25:
In fact I had forgotten some of the solutions, and perhaps move to the Linux operating system is the reason

msoya, 2015-06-26:
I solved my problem. I didn't upload the Marlin software to Arduino before.

I uploaded the Marlin first, then tried to connect from Pronterface, now it is ok;) The connection was succesefully done.

rockstorm101, 2017-12-19:
I'm closing this issue. The root of the problem was found and it was not Printrun related.

closed 2017-12-19 by rockstorm101

adigoe, 2019-08-14:
Open pronterface as administrator. It solved my problem.

Greebeeee, 2020-04-13:
Solution did not work, Thank's for closing issue.

aaronsteers, 2021-01-31:
Thanks, @cron410. Confirmed: closing Cura solved the issue for me.

htrabbit, 2021-03-21:
Thanx CRON410, closing cura solved it for me too!

DivingDuck, 2021-03-21:
An other way is to deactivate (not delete) the USB plugin of Cura.

TRANIER, 2021-03-28:
Résolu, c'est ok après avoir fermé Cura. Merci! Bon print a tous !

mike2moon, 2023-03-04:
simply unplugging the usb from my computer and plug it back in did the trick for me.  It didnt work when i unpluged the printer back and forth but it worked when i unplugged from the computer side.  Maybe it wants the printer connected first before you plug the usb into the computer? hmm

jlopez-cyracom, 2023-07-06:
> I've had this issue a few times, for posterity, make sure you're not running Cura at the same time, or any other slicer that has the ability to print over USB Serial connection. It will steal the COM port and release it once the slicer is closed. Hit connect in Printrun again and it works perfectly.

Can vouch that this was causing the issue for me.