kliment/Printrun #644: \Pronterface\library.zip\pyglet\window\win32\__init__.py:552: UserWarning: Could not set vsync; unsupported extension.

C0p3, 2015-04-01:
Hi All,

Today I plugged the computer onto my 3D printer and, Pronterface is giving me an error message and tells me to check the log--- it connects to the printer but doesn't send any command.

C:\Users\xxxxxxxxxx\Desktop\3D_Pinter\Pronterface\library.zip\pyglet\window\win32\__init__.py:552: UserWarning: Could not set vsync; unsupported extension.

I'm using a vmware fusion/Windows 7 on a PowerBookPro, this computer has two video cards that might be the cause of the error message, the two video cards are the follow - Intel HD Graphics 4000 and NVIDIA GeForce GT 650M.

how can to fix this issue?

thank you
cheers

rhuuk420, 2015-05-22:
I am running into the same issue. unable to run this. it also affects cura UI, so I imagine it has something to do with opengl drivers.

rockstorm101, 2017-12-19:
Hi @C0p3 and @ElJefeDSecurIT, I know the issue is old, but in case you are still using Printrun, does this keep happening with the latest release? I can't reproduce it on my Windows machine.

closed 2018-03-06 by rockstorm101