kliment/Printrun #645: Pronterface for Mac Printrun 2014.03.10 bug

rich1812, 2015-04-10:
Hello,

there is a very annoying bug in this. Perhaps you have overlooked, be seems to me it can be easily fixed.

The issue is that in the Pronterface interface under the Extrude and Reverse buttons, There are two text fields to enter the length of filament. xyz mm@ abc mm/min.

Now in the xyz field, if I enter let's say 120 mm. I quit the program. The next time I start the program, it will retain the 120 mm but it some how adds a decimal point as 120.0mm, now if I try to Extrude or Reverse, I get a error message, "Length is 0, not doing anything." Why in the hell it adds the .0 anyway?
Please do fix this. Thx

kliment, 2015-04-10:
Are you running from binary or from source?

rich1812, 2015-04-10:
I am running the Mac app, so I think it's the binary.

rockstorm101, 2017-12-19:
Hi @rich1812, I know the issue is old, but in case you are still using Printrun, does this keep happening with the latest release?

closed 2018-03-06 by rockstorm101