kliment/Printrun #648: Printrun does not load SD Card

megalomaniac, 2015-04-18:
Hi,
I have a 64 bit Ubuntu 14.04 LTS install.
I can copy gcode files to SD card no problems on my PC but I can't open SD card by clicking the SD icon in printrun to open up, I also tried sudo when running Printrun and another SD card module. I then tried older versions of printrun and still the same problem. Is this printrun of marlin firmware problem? I'm stuck on this problem :-(  
Regards

dlarue, 2015-04-18:
if you are already communicating with your firmware, there would be no need to try using root privileges to talk to the firmware to get data in/out of the SD card.   Do you have your printer printing from the SD card? If so, with the SD card installed in the printer does/can Printrun list/see the file(s) you have on the SD card?

megalomaniac, 2015-04-19:
I'm trying to get my printer to print from the SD card thats attached to the electronics. When SD card is in the printer electronics and I run printrun I can't open the SD card to see the files. Printrun does everything else with no problems, like uploading files from my PC. 

I formated my SD card to FAT32 format does that effect how it's read in the electronics, firmware or Printrun??

dlarue, 2015-04-19:
most of the SD cards are attached to LCD display units, is that the your case?  If so then you first should configure your firmware to enable the SD card so when you copy a gcode file to it on the PC and the insert it, you can use the display and buttons to select it and print from it. Once that is working, then start testing the host software access to the SD card. It can be disabled in the firmware.

kliment, 2015-04-19:
If upload works the SD card is there and works. Can you try it with an
empty card (no folders, no files except the uploaded gcode files) and
enable debug mode in pronterface and tell me what comes back from the
board when you try it?

megalomaniac, 2015-04-19:
I got it to work after I did a upload .gcode file from pc, then the SD card button worked and read the only file in the SD card the .gcode file. I don't understand why I had to first do a upload from PC to get Printrun to read SD card.