kliment/Printrun #649: Extruder FloatSpin boxes missing on Win81

mbird14241, 2015-04-20:
Great tool.

I have been trying to run the binaries I downloaded from Pronterface.com on a Win 8.1 x64 machine (but with 32 bit Python). The extruder length and feed rate FloatSpin fields weren't showing up, they just looked like big pink rectangles.  When I tried to manually extrude some length of filament, it came back with an error because of Marlin's prevent dangerous extrusion setting.  I found that if I updated the version of wxPython to wxPython3.0-win32-3.0.2.0-py27 (as opposed to the 2.8 version suggested on github) it came good.  Hopefully, this saves someone else some time getting it going.

rockstorm101, 2017-12-19:
Hi @mbird14241, I know the issue is old, but in case you are still using Printrun, does this keep happening with the latest release? I can't reproduce it even with wxPython 2.8

mbird14241, 2017-12-19:
Sorry, I disassembled my 3d printer shortly after posting this and haven't
had a chance to get it back together.

rockstorm101, 2017-12-20:
Alright, I'll close the issue then because it seems solved to me. Feel free to reopen if you encounter the same problem again.

closed 2017-12-20 by rockstorm101