kliment/Printrun #650: pronterface losing connection with smoothie after launching sd card print

shoutr, 2015-04-23:
When i connect to my smoothie over ethernet i can control my printer just fine. Then when I try to print from the sd card, i have a message saying the print started and as soon as the temperature is right, pronterface doesn't seem to receive anything from the smoothie anymore. The print continue smoothly. Once it ended, if I want to start another print, I have to disconnect pronterface and reconnect it. I have to do it multiple times, because most of the time the first reconnect end up with a "not currently playing" message spammed in the console. Got this with the previous version and the current one.

kliment, 2015-04-23:
can you switch to debug mode and capture the start and end of an sd
print? That will help me find out what's going on.

shoutr, 2015-04-24:
Here a copy/pasta of the sequence:  (fresh pronterface install on windows 7 64)
Connecting...
SENT: M105
RECV: ok T:21.2 /0.0 @0
Printer is now online.
RECV: Smoothie command shell
SENT: M105
RECV: ok T:21.0 /0.0 @0
SENT: M105
RECV: ok T:21.0 /0.0 @0
SENT: M105
RECV: ok T:21.0 /0.0 @0
SENT: M21
SENT: M20
RECV: SD card ok
RECV: ok
RECV: Begin file list
RECV: backup-firmware/
RECV: documentation/
RECV: drivers/
RECV: models/
RECV: readme.txt
RECV: software/
RECV: source/
RECV: config
RECV: firmware.cur
RECV: windows-files/
RECV: hinge-inner_left.gcode
RECV: y-motor.gcode
RECV: x-carriage.gcode
RECV: x-idler_m5.gcode
RECV: y-idler_603zz.gcode
RECV: foot-front-left.gcode
RECV: contest.gcode
RECV: foot-front-right.gcode
RECV: system volume information/
RECV: foot-rear-lr.gcode
RECV: z-motor_left.gcode
RECV: z-top_left.gcode
RECV: 6mm_bearing_holders2.gcode
RECV: lm6uu bearing support.gcode
SENT: M105
RECV: lm6uu bearing support v2.gcode
RECV: End file list
RECV: ok
RECV: ok T:21.0 /0.0 @0
SENT: M105
RECV: ok T:21.0 /0.0 @0
SENT: M23 lm6uu bearing support v2.gcode
SENT: M105
RECV: File opened:/sd/lm6uu bearing support v2.gcode Size:315457
Print started at: 12:46:23
RECV: File selected
SENT: M24
RECV: ok
RECV: ok T:21.2 /0.0 @0
RECV: ok
SENT: M27
SENT: M105
SENT: M27
SENT: M27
SENT: M27
SENT: M27
SENT: M27
SENT: M27
SENT: M27
SENT: M27
SENT: M27
SENT: M27
SENT: M105
[... 848 lines cut ...]
RECV: Not currently playing
RECV: ok
RECV: Not currently playing
RECV: ok
RECV: Not currently playing
RECV: ok
RECV: Not currently playing
RECV: ok
RECV: Not currently playing
RECV: ok
RECV: ok T:121.8 /0.0 @0
RECV: Not currently playing
RECV: ok
RECV: Not currently playing
RECV: ok
RECV: Not currently playing
RECV: ok
RECV: Not currently playing
RECV: ok
RECV: Not currently playing
RECV: ok
Disconnected.
RECV: Not currently playing

Once the smoothie replied everything, I can print again.

iXce, 2015-04-24:
Do you have the "Enable TCP streaming" option checked ? If so, just disable
it and this issue should disappear, though it may lead to other problems
(non-full buffers)

shoutr, 2015-04-24:
it's disabled

kliment, 2015-04-24:
I'm almost certain Smoothie is ignoring incoming serial commands while
printing from SD, which causes all the other problems.

rockstorm101, 2017-12-17:
Hi @shoutr, what is the status of this issue? Did you find what caused the disconnection?

shoutr, 2017-12-17:
I found nothing to even hint at what caused the disconnection. I got a raspberry and installed octopi on it to run the smoothie from ethernet, been working fine like that since then.

rockstorm101, 2017-12-30:
Alright, I'll be closing this issue then. If anyone else is able to reproduce this problem again please do reopen this issue.

closed 2017-12-30 by rockstorm101