kliment/Printrun #652: Gcode view

ByteRam, 2015-04-29:
When i double-click loaded object Gcode window is showed, i set the layer height and click "Inject Gcode" and nothing appen. Same if i click Edit Layer.
I try in Ubuntu 15 and Windows 8 and log show:

LINUX:
Traceback (most recent call last):
  File "/home/nexus/Printrun/printrun/gviz.py", line 88, in <lambda>
    self.Bind(wx.EVT_TOOL, lambda x: self.p.inject(), id = 6)
  File "/home/nexus/Printrun/printrun/gviz.py", line 208, in inject
    layer = self.layers.index(self.layerindex)
AttributeError: 'dict' object has no attribute 'index'

WINDOWS:
Traceback (most recent call last):
  File "printrun\gviz.pyc", line 88, in <lambda>
  File "printrun\gviz.pyc", line 208, in inject
AttributeError: 'dict' object has no attribute 'index'

SamuelPulpan, 2016-07-31:
ihave got a same problem.

dzach, 2016-11-20:
Same problem here, on kubuntu 16.04 and python 2.7.12

rockstorm101, 2016-12-21:
I get the very same traceback on Debian.

kliment, 2018-03-23:
I believe this has been fixed - can someone verify?

kliment, 2018-03-23:
Sorry, forgot to specify - it should be fixed in the 2.x branch

rockstorm101, 2022-03-04:
I cannot reproduce this anymore. I believe it is fixed, at least for Debian/Ubuntu machines. @DivingDuck could you test on Windows?

DivingDuck, 2022-03-05:
Hi @rockstorm101,
I too can't replicate this for Windows10. I did test it for 2.x RC5 and my version based on RC8 +my changes. 
It looks like this is fixed.
Two injections:
[image]

rockstorm101, 2022-03-05:
Thanks @DivingDuck.

Closing as fixed in the 2.x branch.

closed 2022-03-05 by rockstorm101