kliment/Printrun #653: Connection Error - Serial via Bluetooth2-

bstott, 2015-04-29:
Compiled Pronterface OR direct python versions dated after 1/2015 will not connect to printer on available serial port via bluetooth. Yet, 2014.08.01 readily connects. I've deleted the existing pairing and regenerated but, the new version of Pronterface will not connect. Below is the results:

Connecting...
[ERORR] Could not connect to COM13 at
baudrate 11520:
Serial error: could not open COM13: [Error
1168] Element not found.
 
--- Also, The 2015 version of Pronterface does not read my desktop background and macro settings file...

What information do I need to give you for these issues?

Thanks, as always for a great way to access my printer.

Brian.

bstott, 2016-02-19:
Ping. I tried my setup with changes today, 20160219, and still have the - Bluetooth - unable to connect problem as I'd stated above. The compiled Windows version of pronterface/printrun, 20150315 is not able to connect via Bluetooth serial.
 
My Bluetooth setup will connect with the 20140415 version of pronterface/printrun and also with versions of the competitor - Repeteir. I am also able to readily and easily connect to the printer via an Android Bluetooth connectivity control application.

I do not know where to look to solve this connectivity problem. Any help is appreciated.

kliment, 2016-02-20:
Hey,

This is a really confusing error, as CreateFile (which pyserial calls on windows to connect) is not supposed to be able to return error 1168. Do you have any way of getting on IRC so I can ask you to try a few things?

rockstorm101, 2017-12-17:
HI, @bstott, what's the status of this issue? Did you get it solved? Does it keep happening with the latest release?

closed 2018-03-02 by rockstorm101