kliment/Printrun #654: Printrun sends multiple gcodes when Q:1

vladtcvs, 2015-05-01:
Printrun version 2014.08.01

I have such log of communications:

SENT: M105
RECV: ok Q:1 T:-9 B:-9
SENT: M105
RECV: ok Q:1 T:43 B:-9
SENT: M105
SENT: G91
SENT: G0 X1.0 F3000
SENT: G90
SENT: M114

So, printrun send multiple gcodes, much more than 1.

iXce, 2015-05-01:
That's when jogging right ? G91 puts the machine in relative movement mode, G0 does the movement, G90 resets back to absolute movement, M114 asks for the new position. This looks to be as expected.
Which firmware is that btw ? I've never seen such M105 replies. What does Q:1 means ?

vladtcvs, 2015-05-01:
> This looks to be as expected

I'm expecting, that printrun will wait for answer  for G91, and send G0 X1.0 F3000 only after it.

Firmware is my own.

About Q:1, after I've noticed, that printrun sends 3 gcodes without waiting for answer, I've searched on reprap site about this, and found http://reprap.org/wiki/GCODE_buffer_multiline_proposal. Unfortunately, I don't know, if it is realized or not.

kliment, 2015-05-01:
> I'm expecting, that printrun will wait for answer  for G91, and send G0 X1.0 F3000 only after it.
>
> Firmware is my own.
>
> About Q:1, after I've noticed, that printrun sends 3 gcodes without waiting [...]

plain ok is suppressed in the debug output, but printrun does wait for
an ok after every command. Look at printcore.py for details. As far as I
know you're the first to implement the Q: functionality

rockstorm101, 2017-12-15:
I'm closing this issue because the question was answered.

closed 2017-12-15 by rockstorm101