kliment/Printrun #655: Prontercore - gcoder_line error

drusso1, 2015-05-02:
I have the same problem https://github.com/kliment/Printrun/issues/570. Tried following those instructions, still the same problem. 

WARNING:root:Memory-efficient GCoder implementation unavailable: No module named gcoder_line
Usage: python [-h|-b|-v|-s] printcore.py /dev/tty[USB|ACM]x filename.gcode

Traceback (most recent call last):
  File "/root/Desktop/Printrun/printcore.py", line 56, in <module>
    sys.exit(2)
SystemExit: 2

I'm using pronterface on a raspberry pi running raspbian.

iXce, 2015-05-02:
Well this is definitely not an *error* but just a warning. Do you have cython installed ?

kliment, 2015-05-02:
What is the command you issued immediately before this happened?

drusso1, 2015-05-02:
Yes!
pronsole running the program generates several errors .

drusso1, 2015-05-02:
Idle run or F5

drusso1, 2015-05-02:
Excuse but I decided to switch to windows and I have another problem . I would like to start a printing press of a button created with Tkinter . I would use:
comando = [
    'python',
    '/percorso/a/printcore/printcore.py',
    ' -b %d' % baudrate,
    'COM7',
    /percorso/al/file/g-code
]
subprocess.Popen(comando)

but I haven't  the prontcore.py for Windows.

 could You give it to me ?  think You that works ?

kliment, 2015-05-02:
Instead of using subprocess, import printcore as a module and use that.

drusso1, 2015-05-02:
I print from a GUI in a vending machine where you see the button " Print " .

the user not must configure the printer.

If I launch printcore from Idle ( F5 ) I get this error

Traceback (most recent call last):
  File "C:\Users\scuola\Dropbox\scuola\scuola 2014 15\impresa in azione\sw\Printrun\printcore.py", line 22, in <module>
    from printrun.printcore import printcore
  File "C:\Users\scuola\Dropbox\scuola\scuola 2014 15\impresa in azione\sw\Printrun\printrun\printcore.py", line 20, in <module>
    from serial import Serial, SerialException, PARITY_ODD, PARITY_NONE
ImportError: No module named serial

kliment, 2015-05-02:
> I print from a GUI in a vending machine where you see the button " Print " .
> 
> the user not must configure the printer.

Look at the "using printcore" section in
https://github.com/kliment/Printrun/blob/master/README.md for an example
of how to do that

> If I launch printcore from Idle ( F5 ) I get this error
>
> Traceback (most recent call last):
>   File "C:\Users\scuola\Dropbox\scuola\scuola 2014 15\impresa in azione\sw\Printrun\printcore.py", line [...]

You need to install pyserial

drusso1, 2015-05-02:
I'm sorry you do not know English and I am a beginner . You could write the code ?

kliment, 2015-05-02:
There is a code example in https://github.com/kliment/Printrun/blob/master/README.md in the section titled "using printcore". You can use that code to implement your print button. However, you will need to install pyserial before it will work.

drusso1, 2015-05-02:
I downloaded pyserial-2.7.win32.exe   and when I launch it, I have this message :"No Python installation in the registry". WaY? I Have Python 2.7

drusso1, 2015-05-02:
The code you're talking about is this?
p=printcore('/dev/ttyUSB0',115200)
p.startprint(data) # data is an array of gcode lines
p.send_now("M105") # sends M105 as soon as possible
p.pause()
p.resume()
p.disconnect()

I have to precede it with import printcore?

drusso1, 2015-05-03:
wrong I Something ?
may be:

from printrun.printcore import printcore
p=printcore('/dev/ttyUSB0',115200)
p.startprint(data) # data is an array of gcode lines
p.send_now("M105") # sends M105 as soon as possible
p.pause()
p.resume()
p.disconnect()

drusso1, 2015-05-03:
from printrun.printcore import printcore
p=printcore('COM7',115200)
p.startprint('a.gcode') # data is an array of gcode lines
p.send_now("M105") # sends M105 as soon as possible
p.pause()
p.resume()
p.disconnect()

ERROR:root:Not connected to printer.

printer is COM7 online

drusso1, 2015-05-03:
ERROR:root:Could not connect to COM7 at baudrate 115200:
Serial error: could not open port 'COM7': WindowsError(5, 'Accesso negato.')
ERROR:root:Not connected to printer.

kliment, 2015-05-03:
This means that you do not have permission to connect to COM7. I'm not
sure why this would happen, but a common reason is that some other
program is already using com7 and yo need to close it first.

drusso1, 2015-05-03:
sorry but is very important ...
I restat the pc and there is the same error
with pronsole.py is ok

drusso1, 2015-05-03:
My account is administrator

drusso1, 2015-05-03:
import subprocess
comando = [
    'python',
    'C:\Printrun-master\printcore.py',
    ' -b %d' % 250000,
    'COM7',
    'a.gcode'
]
subprocess.Popen(comando)
[image]

kliment, 2015-05-03:
Contact me on IRC on the Freenode network (my nick is Kliment) and I'll
be able to help you further.

drusso1, 2015-05-03:
I'm daniela

drusso1, 2015-05-03:
I lost you

kliment, 2015-05-03:
When you're connected to IRC you can type
/query Kliment

to open a chat with me

drusso1, 2015-05-03:
No such nick: Kliment

kliment, 2015-05-03:
Try to reconnect - you still appear online to me

drusso1, 2015-05-03:
[image]

drusso1, 2015-05-03:
I lost again  No such nick: Kliment

drusso1, 2015-05-03:
[image]

kliment, 2015-05-03:
This turned out to be a documentation issue - fixed by commit bd99abd

closed 2015-05-03 by kliment

drusso1, 2015-05-04:
I would like to thank you. Yesterday I had a problem with my  INTERNET connection.

parthaero, 2018-03-14:
So is this issue resolved? Cause I checked the commit bd99abd, used the code from there but I’m having the same issue.

kliment, 2018-03-14:
This one is resolved, #842 which you are probably having is not

parthaero, 2018-03-14:
The error which is being mentioned here is exactly what I'm facing. Anyways i'll comment on #842

amrithmmh, 2019-02-15:
I have the same error the command goes to marlin and executes sometimes