kliment/Printrun #656: Command' lines for to set the temperature of the bed and of the extruder

drusso1, 2015-05-04:
I would like to thank you. Yesterday I had a problem with my  INTERNET connection.
 In relation to the same projec thttps://github.com/kliment/Printrun/issues/655 , I need to set the temperature of the bed and the extruder.  are there command lines used?

kliment, 2015-05-04:
> I would like to thank you. Yesterday I had a problem with my  INTERNET connection.
>  In relation to the same projec thttps://github.com/kliment/Printrun/issues/655 , I need to set the temperature of the bed and the extruder.  are there command lines used?

M104 S200 will set the extruder temperature to 200
M109 S200 will set the extruder temperature to 200 and wait until it's
reached before continuing

M140 S200 will set the bed temperature to 200
M190 S200 will set the bed temperature to 200 and wait until it's
reached before continuing

I recommend you add those to the beginning of your gcode file.
You can also send them using p.send_now("M140 S200")
Also, you can put M104 S0 and M140 S0 at the end of your file so the
extruder and bed cool down. For more information on available commands
see
https://github.com/MarlinFirmware/Marlin/blob/Development/Documentation/GCodes.md

closed 2015-05-04 by kliment

drusso1, 2015-05-09:
https://github.com/MarlinFirmware/Marlin/blob/Development/Documentation/GCodes.md

the link is off

kliment, 2015-05-09:
The link was correct but the maintainers have moved the information to a
wiki page which is currently not correctly formatted. Here is the link
to the file before it was removed:

https://github.com/MarlinFirmware/Marlin/blob/d92b52b19bf216eb99e84b05a1ac428fd5fcec98/Documentation/GCodes.md

drusso1, 2015-05-09:
(Y)