kliment/Printrun #657: Cython-based G-Code parser compilation problem

wavemill, 2015-05-04:
Dear,

I tried to compile it but always the same error:
Cython is installed and Gcc also.

root@a20-olimex:/home/olimex# cd /home/olimex/Printrun-master
root@a20-olimex:/home/olimex/Printrun-master# python setup.py build_ext --inplace
WARNING:root:Memory-efficient GCoder implementation unavailable: No module named gcoder_line
running build_ext
building 'printrun.gcoder_line' extension
gcc -pthread -fno-strict-aliasing -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fPIC -I/usr/include/python2.7 -c printrun/gcoder_line.c -o build/temp.linux-armv7l-2.7/printrun/gcoder_line.o
printrun/gcoder_line.c:4:20: fatal error: Python.h: No such file or directory
compilation terminated.
error: command 'gcc' failed with exit status 1

regards,

Wavemill

kliment, 2015-05-04:
Looks like the python development headers are not installed. I'm not
sure what OS you are using but it's normally called python-dev or
python-devel. Try installing that and rebuilding.

wavemill, 2015-05-04:
Dear,

Great thanks, it works!

closed 2015-05-04 by wavemill