kliment/Printrun #659: Reversing Stepper Motor

dEDshort, 2015-05-08:
In repetier host I select invert x-axis for proper operation in print settings.

I would like to try printrun and see no way to invert  steppers in settings.

Am I missing where it is ?

dlarue, 2015-05-08:
I think you are confusing using RepetierHost's EEPROM settings editing(requires RepetierFW ) with Printrun's feature set which does not include editing EEPROM settings. Printrun does allow you to use GCode to change your settings though.

dEDshort, 2015-05-09:
This is in the Repetierhost setup under
 <printer settings>(printer (tab))[Invert direction control(x-axis){check box}.
I am using latest Marlin firmware.
Just wondering if this was possible to do in printrun without having to reflash arduino.

kliment, 2018-03-25:
Sorry, this is not possible in printrun at the moment. We'll keep it in mind for the future but it's usually not a good idea for your printer and your host software to not have the same idea of which way is which - this can lead to confusion when your manual moves go one way but your printed parts don't and you end up with mirrored parts.

closed 2018-03-25 by kliment