kliment/Printrun #662: p=p.printcore('COM7',115200)  NameError: name 'p' is not defined

drusso1, 2015-05-09:
I used this code and everything went well:

from printrun.printcore import printcore
from printrun import gcoder
p=p.printcore('COM7',115200)                    # on Windows
gcode=[i.strip() for i in open('a.gcode')]      # or pass in your own array of gcode lines instead of reading from a file
gcode = gcoder.LightGCode(gcode)
p.send_now("M109 S200")                         # will set the extruder temperature to 200 and wait until it'sreached before continuing
p.send_now("M140 S70")                          # will set the bed temperature to 200
p.startprint(gcode)                             # this will start a print
p.send_now("M104 S0")                           # the extruder cool down
p.send_now("M140 S0")      

suddenly, without changing anything, it appeared this error:

p=p.printcore('COM7',115200) 
NameError: name 'p' is not defined

What could have happened?

kliment, 2015-05-09:
> p=p.printcore('COM7',115200) 
> NameError: name 'p' is not defined
> 
> What could have happened?
that should be p=printcore('COM7',115200)

rockstorm101, 2017-12-15:
I'm closing this issue because the question was answered.

closed 2017-12-15 by rockstorm101