kliment/Printrun #664: Debian Jessie and wxPython

pOmelchenko, 2015-05-17:
Hi! Look at this issues in Cura git

https://github.com/daid/Cura/issues/997

I try replay this solution, but i have trouble.

>omelchenko@envy:~/3dPrint/Printrun$ ./pronterface.py 
>wxPython is not installed. This program requires wxPython to run.
>Traceback (most recent call last):
>  File "./pronterface.py", line 21, in <module>
>    import wx  # NOQA
>ImportError: No module named wx
>omelchenko@envy:~/3dPrint/Printrun$

pOmelchenko, 2015-05-17:
I found the problem why did not happen by the option that is specified in the issues cura. I do not have installed the downloaded packages.

The problem with the launch decided to install a package from a repository

python-wxgtk3.0

rockstorm101, 2017-12-15:
I'm closing this issue because the problem seems solved.

closed 2017-12-15 by rockstorm101