kliment/Printrun #666: Noob User. MakerGear M2 Printer, Mac 10.10.3, Macpronterface errors

elgaotu, 2015-05-21:
HelloAll,

Just got a Maker Gear M2, installed Macpronterface, completed all startup steps suggested by MakerGear, and successfully printed all three files preloaded in the supplied SD card.

When I send any .stl file I've created on my computer to the printer, I get the errors below:

Slicing /Users/elgaotu/Desktop/halbachholder.stl
Running /Applications/Slic3r.app/Contents/MacOS/slic3r /Users/elgaotu/Desktop/halbachholder.stl --output /Users/elgaotu/Desktop/halbachholder_export.gcode
[ERROR] Failed to execute slicing software: 
Traceback (most recent call last):
  File "/Users/craigvanvliet/Desktop/PyInstaller-2.1/pronterface/build/pronterface/out00-PYZ.pyz/printrun.pronterface", line 1257, in slice_func
  File "/Users/craigvanvliet/Desktop/PyInstaller-2.1/pronterface/build/pronterface/out00-PYZ.pyz/subprocess", line 679, in __init__
  File "/Users/craigvanvliet/Desktop/PyInstaller-2.1/pronterface/build/pronterface/out00-PYZ.pyz/subprocess", line 1228, in _execute_child
OSError: [Errno 2] No such file or directory

Loading file: /Users/elgaotu/Desktop/halbachholder_export.gcode
Exception in thread Thread-34:
Traceback (most recent call last):
  File "/Users/craigvanvliet/Desktop/PyInstaller-2.1/pronterface/build/pronterface/out00-PYZ.pyz/threading", line 552, in __bootstrap_inner
  File "/Users/craigvanvliet/Desktop/PyInstaller-2.1/pronterface/build/pronterface/out00-PYZ.pyz/threading", line 505, in run
  File "/Users/craigvanvliet/Desktop/PyInstaller-2.1/pronterface/build/pronterface/out00-PYZ.pyz/printrun.pronterface", line 1372, in load_gcode_async_thread
  File "/Users/craigvanvliet/Desktop/PyInstaller-2.1/pronterface/build/pronterface/out00-PYZ.pyz/printrun.pronsole", line 886, in load_gcode
IOError: [Errno 2] No such file or directory: u'/Users/elgaotu/Desktop/halbachholder_export.gcode'

kliment, 2015-05-21:
You need to put Slic3r in your applications folder

elgaotu, 2015-05-21:
Did the trick, thank you very much!

closed 2015-05-21 by elgaotu