kliment/Printrun #667: I don't get the top menu bar

wbrokow1, 2015-05-22:
I run on a ubuntu laptop with 14.04 LTS.
When I run ./pronterface.py   no top menu appears.
When I run sudo ./pronterface.py  top menu appears 
I am able to connect to the 3d printer and control it, but I want to run without having to use "sudo" everytime.
Please help.
[image]

ntoff, 2015-05-27:
same issue here on OSX 10.6.8

dlarue, 2015-05-27:
the originator posted elsewhere that the cause turned out to be Ubuntu Unity related.

Setting this before starting pronterface.py solved the problem:

export UBUNTU_MENUPROXY=0

ntoff, 2015-05-27:
would that work on OSX though?

dlarue, 2015-05-27:
no, OSX is not Ubuntu. I stated that so it was obvious to people that you and the originator had different issues only similar effects.

JoseJavier616, 2015-07-08:
003491616327809

rockstorm101, 2017-12-15:
Hi @ntoff, did you find a solution for this? or at least the root of the problem?

ntoff, 2017-12-17:
I don't remember, this issue was **over 2 years ago**. I probably learned how2python and fixed it myself or something.

In any case, I stopped using pronterface ages ago and the hard drive in that macbook has been wiped since then so any fix (if there was one) is long gone.

rockstorm101, 2018-03-02:
Alright, I'll close the issue then. Please feel free to reopen if needed.

closed 2018-03-02 by rockstorm101