kliment/Printrun #671: Pronterface will not connect to 3D printer

WilliamFH, 2015-06-02:
Whenever I try to connect my 3D printer I get this error message:

Connecting...
[ERROR] Could not connect to /dev/ttyACM0 at baudrate 250000:
Serial error: could not open port /dev/ttyACM0: [Errno 13] Permission denied: '/dev/ttyACM0'

I have tried varying the baudrate but that has not helped.

Also whenever I open Pronterface this messgae appears in the Terminal: 

WARNING:root:Memory-efficient GCoder implementation unavailable: No module named gcoder_line

i have been trying to crack this all day but have had no success. I am not very experienced with this sort of thing. So any help would be much appreciated.

iXce, 2015-06-02:
definitely looks like an user permission issue. You seem to not have access to /dev/ttyACM0. What does ``ls -lah /dev/ttyACM0`` shows ?

WilliamFH, 2015-06-02:
Thank you very much for the quick response. 
It shows this:

crw-rw---- 1 root dialout 166, 0 Jun  2 22:02 /dev/ttyACM0

iXce, 2015-06-02:
Your user must be in the ``dialout`` group. You can do that using ``gpasswd -a USERNAME dialout``

WilliamFH, 2015-06-02:
what does dialout group mean?
So I type g(my password?) -a (my username?) dialout ?

iXce, 2015-06-02:
it's the group which owns the /dev/ttyACM0 file (root user owns it, dialout group owns it), as show by the output of ls

no, ``gpasswd`` is a command ! so really ``gpasswd -a (myusername) dialout``

dlarue, 2015-06-02:
You need the github pyserial to get 250000 to work. Or accept the possible
issues with a slower but standard speed.

WilliamFH, 2015-06-02:
Ok I tried that command but it was denied:

william@william-XPS-13-2015:~$ gpasswd -a william dialout
gpasswd: Permission denied.

Sorry for being so clueless haha

iXce, 2015-06-02:
Use ``sudo``

WilliamFH, 2015-06-02:
Ofcourse..

after adding sudo to the command it seemed to work and displayed this: Adding user william to group dialout

However when I opened pronterface and tried to connect again I got the same error message as before. I will reboot and try again.

WilliamFH, 2015-06-02:
Ok so after a reboot and another try the error message was replaced by something much friendlier. Thank you very much for your help. 
however the warning still pops up in the Terminal when I open Pronterface, is this smthg I should be worried about? And the Printer isn't responding to any of the commands yet :'D hopefully I will be able to change that.

dlarue, 2015-06-02:
You also have to match the baudrate in your firmware.

WilliamFH, 2015-06-02:
I selected a different baudrate and it worked.
So thank you again everyone!

closed 2015-06-03 by iXce

protist, 2017-09-09:
FWIW in Arch the relevant group is `uucp`. You can work this out by looking at the group owner for the device, i.e. `ls -l /dev/ttyACM0`.