kliment/Printrun #673: Objects printed in the air when multiple objects printed one by one

gege2b, 2015-06-08:
Hi all

I got a problem with a gcode file containing 6 identical objects

I used Slic3r to make the Gcode, checking the "complete individual object", so first object would be printed completely, then the second, etc...

When printing, the first object is printed correctly, BUT the second object begin at the max Z of the first object (the Z axis doesn't move back to Z0.2)

The Gcode looks good ( https://dl.dropboxusercontent.com/u/34934879/20150601-165554-logo_olcq_0.2-20.gcode ) and Printrun preview is good too (each object appears one by one and on the plate when moving the Z slider)

So it's obviously not a Slic3r problem

What I don't really know, is if it's a Printrun problem or a Marlin one (a rather recent build, just before they begin tagging their release)

I'm using commit 60ed409013892c4c66440cd6b7a56d3d63e9ee0e on win7 x64

If it's not a printrun problem, I'll copy/paste this on marlin's tracker ^_^

cheers

iXce, 2015-06-08:
As far as I know Printrun only sends whatever gcode is given to it, so it
sounds a bit unlikely that it's Printrun's fault.
I just loaded your gcode in Printrun and printed it with Printator as a
printer simulator, and it looks perfect.

kefir-, 2015-06-08:
The move back down to the build plate for the next part looks like this:

    G1 Z0.200 F7200.000

It's moving down from Z6.0 which was the previous setting. F7200 seems fast, do you have a Delta printer? If not, I'd suggest looking into if this speed works. Try manually homing and then sending move commands like:

    G1 Z8 F7200
    G1 Z2 F500
    G1 Z8 F1000

etc... to move up and down some distance, and play around with speeds to see what works and what doesn't. It could be a slic3r bug if the speeds are too high.

gege2b, 2015-06-08:
I was precisely looking at the speed right now, and then I see your post
In fact, the speed seems to high for my printer Z axis ! 

It's using Slic3r's travel speed (mine is set to 120mm/s - OK for XY but not for long Z moves)
I have an old Prusa Mendel, but not at the office, so I'll try this after work if I can

I haven't thinked about that at first, and it's certainly what's causing this issue
I'll certainly set a lower Z speed limit in Marlin

thanks both of you for your help

closed 2015-06-08 by gege2b

iXce, 2015-06-08:
Glad it's solved ! Z speed issue can be confusing as hell indeed !

gege2b, 2015-06-24:
Hi there

I've modified my marlin config to set max z feedrate to 2mm/s and indeed, I can confirm this was the problem, my printer cannot go faster on the Z axis

The reason I haven't noticed this before is I've never done a long Z move on a print, I think Z acceleration was saving my prints until now ^_^

So this was neither a  printrun nor a marlin issue, just a dumb operator (the more common bug is between the chair and the machine, isn't it ?)

kefir-, 2015-06-24:
Interesting. Should slicers expect the firmware to cap the speed and just always max out the Z moves, or should Slic3r have a different speed setting for Z travel? And perhaps also Z acceleration? These could maybe be placed under Printer Settings to avoid cluttering the Speed configuration window, and I think it'd make sense to have different settings for different printers.

gege2b, 2015-06-24:
I'm not sure
I guess leaving the firmware caping the max Z feedrate would avoid another setting on the slicer's side (they already have a lot)

Normally, the firmware is tweaked for the hardware, I think the speed limitation should be done in the firmware

In fact, I don't know if other firmware are caping the feedrate like Marlin does ?
If they don't, so yes, slicers should probably add a speed limit setting for Z axis