kliment/Printrun #674: length, speed number and buttons not showing up, limited function

nicksears, 2015-06-11:
[image]
I'm having a weird issue on ONE of my computer/printer pairs. (Windows 8.1x64, Printrun-Win-Slic3r-03Feb2015)
When I open pronterface the length/speed numbers and buttons are "greyed" out (more like a beige-ish color).
I can still click on them, type "1" for example and hit extrude, and it will extrude 1mm (when at temp) and confirm in the terminal.
However, this is very difficult to use when I can't increment up/down or change the length/speed easily.

I've tried downloading a new copy, deleting the old config file in my documents, etc, but have found no solution yet. Other computers are also windows 8.1x64, with the same printers, but they work fine. Is it something to do with python dependencies? I'm using the precompiled version from (http://koti.kapsi.fi/~kliment/printrun/)

kliment, 2015-06-11:
I've never seen anything like this before. Honestly have no idea where
to start in figuring it out. Is there anything at all that is different
about this computer compared to the others?

nicksears, 2015-06-11:
Darn. No I have no idea. I might have run the version that I had to compile myself at one point, so did I maybe install slightly different versions of python that are not playing nice? I'm not great at this stuff, so I don't know if it can even work like that.

Also, I've found that the (Printrun-Win-Slic3r-10Mar2014) version works, so at least I can go with that for now.

wiganite, 2015-07-07:
Glad I found this.  I've also got exactly the same issue will try the mar2014 version.

rockstorm101, 2017-12-15:
Hi @nicksears, does this keep happening with the latest release?

nicksears, 2018-02-20:
Unfortunately I'm not working with that computer anymore and cant say. New version seems awesome on my other computers though :)

closed 2018-02-20 by nicksears

Regaor, 2020-04-24:
same problem here... has anyone solved?
It worked very well and suddenly the problem appeared.
I'm in Widows 8.1. Thank you

[image]

Regaor, 2020-04-24:
a working solution for this was given in #856