kliment/Printrun #675: Feature request: Option to load printrunconf from different folder (mobile version)

Chrisu02, 2015-06-14:
Hy

would be cool if there would be a option to be able to change the path of the configuration file.
this would help to get running a mobile (usb-thumbstick) or network -version running of it.

something like the --datadir command at slicer

Chri

dlarue, 2015-06-14:
as in using the -c CONF or --config CONF or --conf CONF options?

./pronterface.py --help
usage: pronterface.py [-h] [-v] [-c CONF] [-e EXECUTE] [-a] [filename]

Printrun 3D printer interface

positional arguments:
  filename              file to load

optional arguments:
  -h, --help            show this help message and exit
  -v, --verbose         increase verbosity
  -c CONF, --conf CONF, --config CONF
                        load this file on startup instead of .pronsolerc ; you
                        may chain config files, if so settings auto-save will
                        use the last specified file
  -e EXECUTE, --execute EXECUTE
                        executes command after configuration/.pronsolerc is
                        loaded ; macros/settings from these commands are not
                        autosaved
  -a, --autoconnect     automatically try to connect to printer on startup

Chrisu02, 2015-06-14:
Oh, thanks, i didn`t found the infos at the readme.md  that will help me :)

closed 2015-06-14 by Chrisu02