kliment/Printrun #676: Preset for Temperatures does not affect Selection on the main screen  02/2015 ver.)

Chrisu02, 2015-06-14:
I just noticed that when i change the temperatur values for bed/extruder ABS/PLA at the "Printer Settings" it does not change at the main screen.
there are always the 185/230  60/110 values.

at the older 03/2014 it does work normally.

rockstorm101, 2017-12-15:
Confirmed.

 * OS: Windows 10 Pro
 * Printrun Version: 1.6.0

xoan, 2018-04-01:
Still present in 2.0.0rc5

az143, 2018-11-07:
as far as i can tell from poring over the code of the 1.6.0 debian version (i've also checked current head, same issue but line 368),
it's line 349 in settings.py that looks for callbacks named '__whatever_cb', but what was actually setup for the temps reading from rc file is '_whatever_cb', ie. just one underscore.  with that one character change i'm finally seeing the expected configured temperatures in the dropdown menus

```
--- settings.py.orig
+++ settings.py
@@ -346,7 +346,7 @@
         try:
             cb = None
             try:
-                cb = getattr(self, "__%s_cb" % key)
+                cb = getattr(self, "_%s_cb" % key)
             except AttributeError:
                 pass
             if cb is not None: cb(key, value)
```

rockstorm101, 2020-05-08:
Closing. Fixed by #971. Thanks a lot.

closed 2020-05-08 by rockstorm101