kliment/Printrun #677: Mac OS X Yosemite - GUI problem

pOmelchenko, 2015-06-17:
The problem with the graphical user interface in Mac OS X Yosemite

I downloaded a binary file from http://koti.kapsi.fi/~kliment/printrun/

And when you start the application I see here a picture
[image]

rockstorm101, 2017-12-15:
Hi @pOmelchenko, does this keep happening with the latest release?

pOmelchenko, 2017-12-15:
Heh, I thought this project was dead. For a minute, I check

pOmelchenko, 2017-12-15:
[image]
[image]

The problem is that for 2 years since the discovery of this problem, I have already updated my system several times. Current release preserves the urgency of the interface problem

pOmelchenko, 2017-12-15:
[image]

rockstorm101, 2017-12-15:
Judging by the screenshots in https://github.com/kliment/Printrun/issues/677#issuecomment-352112720 it looks good enough to me. But then in https://github.com/kliment/Printrun/issues/677#issuecomment-352113473 I guess the problem persists?

pOmelchenko, 2017-12-15:
Yes, the problem remained. For some reason, on the screenshots, black became white, but not the color of the main background

kliment, 2017-12-15:
This is a wx issue which can only be fixed by moving to a newer wx version - this will have to be done eventually but will take a fair bit of effort. Thanks for your report.

a2k-hanlon, 2021-02-02:
Wow this has been open for a long time. I'm going to close it since it's so out of date. @pOmelchenko if you still use Pronterface, please download a recent release from https://github.com/kliment/Printrun/releases/latest and feel free to open a new issue if you run into any issues.

closed 2021-02-02 by a2k-hanlon

pOmelchenko, 2021-09-07:
I checked. That problem is fixed, all looks good. Thanks