kliment/Printrun #679: Unusual error: "kKш"

arkcol, 2015-06-24:
Hello! I have RepRap Prusa i3 and i use Pronterface ver. 10 Mar 2014.
During long printing my 3D-printer stops stepper motors and i have strange string in pronterface's console: "kKш". For resuming printing i need press "Pause" and after that  press "Resume".
[image]
Why do i have this error?

rockstorm101, 2017-12-15:
Hi @arkcol, did you find a cause for this message? Has it ever happened again?

arkcol, 2017-12-15:
Hello! No i have not found a reason of this.

rockstorm101, 2017-12-17:
Does it keep happening? Are you able to reproduce it?

arkcol, 2017-12-17:
No, i can't reproduce it.
This issue was 2 years ago.

rockstorm101, 2017-12-19:
Alright, I'll close the issue then. Please reopen it if this happens with the latest version.

closed 2017-12-19 by rockstorm101