kliment/Printrun #682: much of  requirements.txt not satisfiable in current Linux releases (Ubuntu 15.04 x64)

AndKe, 2015-06-28:
At best, execution looks like this: 

WARNING:root:Memory-efficient GCoder implementation unavailable: No module named gcoder_line
WARNING:root:psutil unavailable, could not import power utils:No module named psutil
[ERROR] 3D view mode requested, but we failed to initialize it.
Falling back to 2D view, and here is the backtrace:
Traceback (most recent call last):
  File "/home/andre/RepRap/Printrun-Jun15/printrun/gui/viz.py", line 70, in __init__
    import printrun.gcview
  File "/home/andre/RepRap/Printrun-Jun15/printrun/gcview.py", line 22, in <module>
    from .gl.panel import wxGLPanel
  File "/home/andre/RepRap/Printrun-Jun15/printrun/gl/panel.py", line 27, in <module>
    import pyglet
ImportError: No module named pyglet

iXce, 2015-06-28:
pyglet is at least in APT: http://packages.ubuntu.com/vivid/python-pyglet
psutil is definitely in pypi https://pypi.python.org/pypi/psutil (though maybe a version newer than what we support) ?
gcoder_line is a custom module we ship (see README.md on how to compile it)

iXce, 2015-06-28:
psutil in the version we support is in APT for 15.04: http://packages.ubuntu.com/vivid/python-psutil

AndKe, 2015-06-29:
ok, so my report should say the requirements leave a lot of guesswork.
I suggest it's reformatted as: 
Linux requirements:
sudo apt-get install python-pyglet python-psutil ....

It's hard to guess the package names otherwise.

iXce, 2015-06-29:
Everything is definitely on pypi, so pip -r requirements.txt should work great, no ?
Distribution-specific tools like apt-get cannot seriously be put as "linux requirements" ^^

AndKe, 2015-06-29:
I must have missed pip completely (thing I followed readme.md)
Of course, I do not mean to include apt-get among tools, just tried to illustrate use of actual package names.

iXce, 2015-06-29:
As far as I can tell it's in the readme ^^ https://github.com/kliment/Printrun#running-from-source

AndKe, 2015-06-29:
yep - you are right.

closed 2015-06-29 by AndKe