kliment/Printrun #684: Pronterface will not let me change extrusion amounts

bbartle, 2015-07-09:
Hi, 

I'm trying to calibrate my E-Steps and pronterface won't let me change the extrusion amount. It would originally extruded a preset 5mm when prompted, but changed to 1000mm for some reason. The software says 1000mm is too long to extrude and since it doesn't let me enter a new extrusion value, I cannot manually extrude. I also tried commands such as G1 E5 and M101 E5 which yielded no results.

Please let me know if you have any suggestions or know of any other G or M codes I could use to manually extrude. I'd greatly appreciate the help.

Thanks,
Bryan

ntoff, 2015-07-22:
precompiled binary or running from source?

bbartle, 2015-07-22:
it's precompiled

ntoff, 2015-07-22:
the extrusion length is stored (along with all the settings) in .pronsolerc under "last_extrusion"

.pronsolerc is in your user folder, /home/username on *nix or c:/users/yourname on windows 7

Shahriar57, 2016-06-17:
I have same issue , Printrun 2014.08.01 , i thibk it is source . Length and speed filed are red and can not fillup them , also when i click on Extrude or Reverse got "Extruding 1000.000000mm of filament.
echo: too long extrusion prevented " !!! can you help please to resolve it

jam3sward, 2016-11-30:
I ran into this problem on Windows 10,  found the settings in C:\Users\YourName\printrunconf.ini, changed the value from "set last_extrusion 1000" to "set last_extrusion 10" and that worked. The Length and Speed fields are still red and can't be edited.

Kanguros, 2016-12-14:
I'm experiencing the same problem with printrun version 2014.08.01 and windows 10. On windows 8.1 works fine.

jot4p, 2018-01-15:
Hello, 

I have the same problem.. the Length and Speed fields are still red and can't be edited.

rockstorm101, 2018-02-14:
Hi, a working solution for this was given in #856.

closed 2018-02-14 by rockstorm101