kliment/Printrun #685: Pronterface manual contols, gauges, SD Printing, other issues

Tdawg069, 2015-07-21:
Hi. I've noticed the following issues when using Pronterface (ver. Feb 2015 - Windows):
- x,y,z controls are absolute, not relative. While I'm sure there is a gcode that can be sent to change this, I feel the x,y,z contols should always be in "relative" mode, as they are labelled "+x", "-y", "+z", etc. and not "x={user value}", etc. It does however work in Relative mode once the "center-x,y" button is pressed. There is also no indication of its current position.
- x,y controls cannot be used without altering the z position. Pronterface does not allow you to click on just the arrows without highlighting a z-ring value. Even "center-x,y" causes a shift in z.
- There is only one nozzle gauge which is linked to tool 0. Even if the tool in use is changed to tool 1, the gauge is still linked to tool 0. Also, for dual-colour printing, there should be an additional gauge seen. Same for Tri-colour, etc.
- SD Printing is not tracked at all! There is no diagram (2D or 3D) and the time estimates are way off. a 2 hour print will show as several days remaining.
When a dual-colour gcode file is loaded, there is an offset in the colours. I assume this is due to the offset that Cura applies to the gcode, as set in the machine settings. Pronterface should have a similar setting which caters for such offsets, in order to draw the objects correctly.
- Often when the computer goes on standby, then wakes, I can no longer connect to my printer (says device malfunctioned). However the device continues printing.*
- When my laptop power cable is plugged-in or unplugged, however, printing stops completely and cannot continue. Also shows error message of device malfunctioning.*
- Perhaps also include a basic gcode help file, which would help for manual movement and maintenance of the machine.

*These may be hardware related. I am currently using a R3 Robobeast (RepRap-based). A fairly new company in South Africa which sells custom 3D Printers

Hope these issues are taken into account when developing the next version of Pronterface.

Regards
Tdawg069

ntoff, 2015-07-22:
Not sure what you mean about not being able to select an x or y position without Z, I don't see that issue.

The controls also seem to be in relative mode for me

Printing from SD means the firmware is reading the file directly off the SD card, not really any way for pronterface to track that unless the firmware on the printer sends some kind of feedback I would imagine.

>When my laptop power cable is plugged-in or unplugged, however, printing stops completely and cannot continue. Also shows error message of device malfunctioning.*

I'm going to assume that's something specific to your laptop because I don't see that issue at all. There's a chance windows is turning off the USB ports to save power. Check your power saving options.

rockstorm101, 2017-12-15:
Hi, @Tdawg069, what's the status of this issue? Were you able to solve your problems?

closed 2018-03-02 by rockstorm101