kliment/Printrun #687: Upon windows 10 update pronteface crashes and leaves heaters on

jbalat, 2015-07-26:
Not sure what you can do to turn off the heaters but this almost caused a fire. I have turned off windows automatic updates but it happened again

JoseJavier616, 2015-07-26:
003491616327809

JoseJavier616, 2015-07-26:
De José Javier alcohol para crimen printrun crimen pinzan Climent Climent
Climent England

003491616327809

kliment, 2015-07-26:
Hello @jbalat. Do you get any error messages when the crash happens? Is your computer going to sleep or suspending when this happens?

zohargolan, 2016-10-17:
Hi, 

I know this problem is from a treat ago, but lately I started seeing the same thing in my windows 10 machine. This was at a series of windows update. Any resolution to this problem?

rockstorm101, 2017-12-15:
Hi @jbalat and @zohargolan, what is the status of this issue? Does this keep happening to you?

jbalat, 2017-12-16:
Since then I now write my gcode to the sdcard and print from sd. I can’t trust windows not to do an update.

rockstorm101, 2018-03-02:
I'm closing this issue then. Please feel free to reopen if needed.

closed 2018-03-02 by rockstorm101