kliment/Printrun #688: Printrun unable to open ngc files

hhanff, 2015-08-13:
Hi!

I want to use my self-built 3D printer as a CNC mill.
I'm using pycam as the toolpath generation software. After having generated the toolpath with pycam, I can verify the paths in openSCAM. 
https://gist.github.com/hhanff/4b3d8ad5d39bc9383e1f

Messages from Printrun are:
Loading file: /home/hhanff/Dokumente/Privat/Hobbies/3D_Print_ABS/models/The_Essential_Calibration_Set/0.5mm-thin-wall(0).ngc
[WARNING] raw G-Code line "S1000.00000" could not be parsed
Exception in thread Thread-28:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 810, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 763, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/home/hhanff/Dokumente/Privat/Hobbies/3D_Print_ABS/software/printrun/printrun/pronterface.py", line 1456, in loadviz
    assert(generator.next() == next_layer)
AssertionError

[WARNING] raw G-Code line "S1000.00000" could not be parsed
[WARNING] raw G-Code line "S1000.00000" could not be parsed
Loaded /home/hhanff/Dokumente/Privat/Hobbies/3D_Print_ABS/models/The_Essential_Calibration_Set/0.5mm-thin-wall(0).ngc, 3733 lines
0.00mm of filament used in this print
The print goes:
- from 0.00 mm to 0.00 mm in X and is 0.00 mm wide
- from 0.00 mm to 0.00 mm in Y and is 0.00 mm deep
- from 0.00 mm to 0.00 mm in Z and is 0.00 mm high
Estimated duration: 0 layers, 0:00:00

But when I open the generated ngc file in pronterface, I cannot even see a model in the preview area. 
It seems that pronterface is not capable of processing ngc files. Is that true?
Where is my mistake?
Any suggestins?

Greetings,

Hendrik

iXce, 2015-08-13:
It seems we don't support the line "S1000" in the gcode file. What is it meant to do ?

hhanff, 2015-08-13:
S: Set Spindle Speed.
Good hint. I will try commenting out that line because for the moment I can only set the spindle speed manually

hhanff, 2015-08-13:
I did 
```
cat 0.5mm-thin-wall\(0\).ngc | grep -v S1000 >0.5mm-thin-wall\(1\).ngc
```
to filter out the S1000 and the warning disappeared.
Nevertheless, I cannot see the object in the preview area and the message that the object has 0mm in all directions remains

```
Loading file: /home/hhanff/Dokumente/Privat/Hobbies/3D_Print_ABS/models/The_Essential_Calibration_Set/0.5mm-thin-wall(1).ngc
Loaded /home/hhanff/Dokumente/Privat/Hobbies/3D_Print_ABS/models/The_Essential_Calibration_Set/0.5mm-thin-wall(1).ngc, 3730 lines
0.00mm of filament used in this print
The print goes:
- from 0.00 mm to 0.00 mm in X and is 0.00 mm wide
- from 0.00 mm to 0.00 mm in Y and is 0.00 mm deep
- from 0.00 mm to 0.00 mm in Z and is 0.00 mm high
Estimated duration: 0 layers, 0:00:00
```

hhanff, 2015-08-13:
A more general question: Is printrun capable of controlling CNC milling machines, or is there another software which suits better?

dlarue, 2015-08-13:
the go-to open source software for CNC and milling machine control is
Machinekit and its father, LinuxCNC. LinuxCNC was created by NIST over a
decade ago and Machinekit was forked from it a few years ago.

iXce, 2015-08-13:
well, printrun is designed for extrusion printer with an E axis which extrudes plastic, this is why nothing appears: you have no extrusion

I checked your file, and I think this is not valid gcode syntax, we expect all moves to be preceded by a G tag, and you have multiple moves for a single G tag there.

For CNC I'd go for LinuxCNC for instance !

hhanff, 2015-08-13:
Thanks IXce. I also saw LinuxCNC. The drawback is, that I would need a computer with a physical parallel port which I don't have (USB to parallel is not an option as far as I know)
Furthermore my Printer/CNC mill has an Adruino running Marlin, and I could not find out if that can receive its commands from LinuxCNC.

dlarue, 2015-08-13:
I thought Marlin was designed for 3D printers. If you think you have Marlin
working for your CNC mill then really you don't need much of a UI if your
gcode is correct or do you?  So it should be all about your slicer
generating your gcode. Then maybe you can use something like OctoPi or
something to just shovel the gcode to the arduino/Marlin.

FYI, there is the Beaglebone Black which might work for the CNC machine. I
know it runs Machinekit because it's being used to run one of my 3D
printers. Arduino/Marlin runs the other.

hhanff, 2015-08-14:
It seems I have a workflow now which does at least not produce any errors any more and visualizes the toolpath in Pronterface:
1. PyCam for Toolpath generation
2. PycamToGCodeConverter.jar to Convert/correct the GCode
3. Pronterface
4. Marlin/Arduino

Thanks for the hints and the discussion.

Hendrik

rockstorm101, 2017-12-15:
I'm closing this issue because the question seems solved. Feel free to reopen if you think otherwise.

closed 2017-12-15 by rockstorm101